sidoh/esp8266_milight_hub #328: After some time of operation, no MQTT states topics are emitted any more

HenkDemper, 2018-09-03:
Hello Chris. The 2nd issue I encounter when using V1.8.1 of the Gateway that after some usage (days, weeks ?) there are no MQTT state topics emitted anymore, only updates. For instance for group 1-4 using a FUT006 remote I then only see the .../update/.../1-4 etc MQTT topics, but no .../states/.../1-4 ones. Command topics are working Ok during this state. As part of my openHAB automation (keeping the items in sync) depends on correct states messages, this is now broken...

When I restart the MiLight hub, it directly emits both MQTT update and states topics again. It seems like a resource problem or some internal deadlock/timeout. I am happy to debug something specific when it happens again (I noticed it about 3 times now). Thanks for any help.

sidoh, 2018-09-03:
Are you able to collect serial logs?

HenkDemper, 2018-09-03:
Not in my current setup. I will need to compile myself and keep attached to always-on computer (and wait till it happens) likely to achieve this ? Any instruction how to setup ?

sidoh, 2018-09-03:
A bit here:

https://github.com/sidoh/esp8266_milight_hub#setting-up-the-esp

Compile flags are in `platformio.ini`.  The relevant flags are `MQTT_DEBUG` and `STATE_DEBUG`.

I don't have updates enabled in my setup, but I've been running 1.8.1 since it was released, and haven't noticed interruptions to the state topic.

HenkDemper, 2018-09-03:
...and if I make that build, then that would always log serial -> USB -> (always-on) computer, right (no logging possible over WiFi / Web GUI or alike) ?

sidoh, 2018-09-04:
Correct.  Unfortunately no way to receive serial logs except through a computer the ESP8266 is attached to.

Agreed it doesn't feel clean.  This definitely shouldn't be happening.

HenkDemper, 2018-09-04:
Maybe the bug you found in issue #327 also fixes this issue (after all, if memory gets corrupted, anything can happen) ? I think it is best I wait for the #327 fix, then try if this issue #328 still exists...

sidoh, 2018-09-05:
Totally plausible.  Do you use group 0 anywhere in your setup?

sidoh, 2018-09-05:
Got it.  Then yes, this could totally be the issue.  Hopefully I'm able to get a fix out soon =)

sidoh, 2018-09-08:
@HenkDemper how's it been looking so far?

HenkDemper, 2018-09-09:
I haven't used the lights via remote / automation a terrible lot the past 4 days, but I just tried to break it by frequently turning them On/Off via the remotes group 1-4 and 0 and I keep getting the correct MQTT topics for both updates as well as states, so looking good so far 👍

sidoh, 2018-09-10:
Cool.  I'm going to close in hopes that this was the issue.  Please feel free to re-open if you notice it again.

closed 2018-09-10 by sidoh

HenkDemper, 2018-10-31:
Hi Chris. Unfortunately I'm noticing the issue is still appearing: I don't get any state changes anymore, while I do get update changes (both state and update changes are configured). For instance:

```
>mosquitto_sub -v -t milight/#

milight/commands/0x887E/rgb_cct/1 {"state":ON}
milight/updates/0x887E/rgb_cct/1 {"state":"ON"}
milight/commands/0x887E/rgb_cct/1 {"state":OFF}
milight/updates/0x887E/rgb_cct/1 {"state":"OFF"}
```

I was running v1.8.2 of the Gateway, just updated to latest v1.8.5, but given the changes doubt that this will help... Because of the reboot it now works again, for the same ON & OFF commands we now get:

```
milight/commands/0x887E/rgb_cct/1 {"state":ON}
milight/updates/0x887E/rgb_cct/1 {"state":"ON"}
milight/states/0x887E/rgb_cct/1 {"state":"ON","level":100,"kelvin":65,"bulb_mode":"white"}
milight/commands/0x887E/rgb_cct/1 {"state":OFF}
milight/updates/0x887E/rgb_cct/1 {"state":"OFF"}
milight/states/0x887E/rgb_cct/1 {"state":"OFF","level":100,"kelvin":65,"bulb_mode":"white"}
```

Will keep monitoring the issue...