sidoh/esp8266_milight_hub #331: Cannot unpair device

UnReal933, 2018-09-05:
I paired 4 GU 10 RGB-CCT Mi-Light (FUT103) to one ID by mistake and they're now controlled by one ID. Sadly I can't unpair them, the "unpair" button just don't work.

PetricaM, 2018-09-06:
In order to unpair, you need to (completely) cut the power to all bulbs then turn it on and quickly press the unpair button with the same remote ID the initial pairing was done (also, ensure the appropriate remote type is selected in the gateway’s webpage - for FUT103 can use RGB+CCT or FUT089). I think it would be easier to do this with each bulb individually. Unless all bulbs are in the same location and a breaker can control them all,  it would be a challenge to turn all of them on and off in the same time (due to range aspects some of them might not receive the signals from gateway; FUT103 do not have the best thermal design and, case they get hot, they tend to lose packets).
Anyway, unless forced by a specific set of circumstances, you can keep the current pairing and then pair the bulbs with a new remote ID (although the pairing and unpairing procedures are the same in what concerns the bulbs) as each bulb can be associated with up to four different remote IDs. Unpairing does actually clear the bulb to factory settings rather that forgetting a single remote ID (it deletes all the associated remote IDs the bulbs have used).

poudenes, 2018-09-06:
And remember. When the bulb is also paired to other device. It will unpair fully. So you have to pair them with every device again. (i figured that out last day)

UnReal933, 2018-09-08:
Thanks you, I reinstalled the Hub and followed your advices, it's working now, thanks !

closed 2018-09-08 by UnReal933