sidoh/esp8266_milight_hub #336: Domoticz MQTT Protocol implementation

poudenes, 2018-09-11:
Hi All,

Everything work great. Maybe a implementation of the Domoticz MQTT would be nice to get a close system.

When you turn on bulbs inside the WebUI the bulbs in domoticz won't get on status. When you send a MQTT signal then Domoticz can turn on the bulb as well (reverse feedback)

https://www.domoticz.com/wiki/MQTT

sidoh, 2018-09-11:
Perhaps I'm missing something, but I think this is already implemented:

https://github.com/sidoh/esp8266_milight_hub#updates

The format is optimized for HomeAssistant's `mqtt_json` component, but if it doesn't work natively with Domoticz, it should be easy to morph it into something that does.

poudenes, 2018-09-11:
I tried the MQTT already but indeed its not work correct. Thats why i post the link also :)

sidoh, 2018-09-11:
I'm not sure how I feel about baking in native support for every home automation platform under the sun.  My tack has been to have one format that's complete enough to work with any platform.

I think we have several Domoticz users here.  Perhaps someone could provide pointers on how they have it set up?

@joshdinsdale raised a similar issue in #186.  Josh, did you happen to get espMH MQTT working with Domoticz?  If so, could you share how you set it up?  Would be a great addition to this project's wiki. :)

closed 2018-09-17 by poudenes