sidoh/esp8266_milight_hub #337: 1.8.1 to 1.8.2 gives lag again

poudenes, 2018-09-13:
Hi All,

Updated to 1.8.2, reboot Module and Domoticz. Checked all settings but have a big lag back again.
When i revert back to 1.8.1 then everything is ok gain.

sidoh, 2018-09-13:
It's really unlikely that changes in 1.8.1 -> 1.8.2 caused anything like this.  Changelog here:

https://github.com/sidoh/esp8266_milight_hub/pull/335

I'm happy to help, but this is probably not something I'm going to look into myself without a bit more detail from you.  For starters, a concrete way to reproduce it. =)

poudenes, 2018-09-13:
This is what i did:

1) Have a good working config. with 1.8.1. Here the .Json config

`{"admin_username":"","admin_password":"","ce_pin":16,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":25,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"","mqtt_username":"","mqtt_password":"","mqtt_topic_pattern":"","mqtt_update_topic_pattern":"","mqtt_state_topic_pattern":"","discovery_port":48899,"listen_repeats":3,"state_flush_interval":20000,"mqtt_state_rate_limit":800,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":1000,"packet_repeat_minimum":5,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Slow toggle","led_mode_packet":"Flicker","led_mode_packet_count":3,"device_ids":[43681,43682,43683,43684,43690,43691,43692,43693,43694,43695,43706,43707],"gateway_configs":[[43690,5980,6],[43691,5981,6],[43692,5982,6],[43693,5983,6],[43694,5984,6],[43695,5985,6],[43681,5986,6],[43682,5987,6],[43683,5988,6],[43706,5989,6],[43684,5990,6],[43707,5991,6]],"group_state_fields":["state","status","brightness","level","hue","saturation","color","mode","kelvin","color_temp","bulb_mode","computed_color","effect","device_id","group_id","device_type"]}`

2) Updated the firmware on the flow (using the upgrade function in webGui.

3) reboot the config (Pull out power and turn back) also reboot Domoticz Config.

4) Did some tests with switches and Domoticz will send 2 commands:
- Turn on bulb
- SetKelvin and Dim level

The first command work, second take long time before it work or even nothing, then i see a error msg inside Domoticz Log:

`2018-09-13 10:30:34.163 Error: AppLamp: Error sending command to Bridge!...`

When i flash back to 1.8.1 and reboot config and domoticz again, then everything is normal again.

So i cant find anything that will cause this lag.

With 1.8.1 all bulbs work correct RGBWCCT and RGBW

poudenes, 2018-09-15:
maybe stupid question. how can i do a debug on the ESP log? Sorry im not so known with the technical background of ESP....

sidoh, 2018-09-15:
I use platformio to view serial logs.  The command I use is `pio device monitor`.

You'll have to compile with the aforementioned flags enabled (edit in `platformio.ini`).

sidoh, 2018-09-15:
also - I'm not able to reproduce this manually sending UDP packets.

poudenes, 2018-09-17:
tried it, but still dont know how to do this... will try later

poudenes, 2018-09-23:
Maybe a idea to have debug log option in web ui. I tried everything but cant get it work to get it work with platformio.

sidoh, 2018-09-26:
@poudenes i like the idea in the abstract, but the reason it's done this way is to remove overhead.

Here's a compiled binary:

[firmware-1.8.3-DEBUG.bin.gz](https://github.com/sidoh/esp8266_milight_hub/files/2418349/firmware-1.8.3-DEBUG.bin.gz)

poudenes, 2018-09-26:
@sidoh i will try this compiled one. And still have use platformio to get the debug log? Cleaned my Mac and install everything clear. Hope platformio will work now.. Get some error while installing.

sidoh, 2018-09-27:
I use platformio with the command I included above.  There are lots of programs to monitor serial logs, though.  The Arduino IDE has a serial monitor, for instance.

poudenes, 2018-09-27:
Here a debug log. For me it seems like "hdjskfhkjsd" hope you can figure out why i have lags with 1.8.2 and 1.8.3 and not in 1.8.1

`[MiLightUdpServer port 5989] - Handling packet: 20 00 00 00 16 02 62 3A D5 ED A3 01 AE 08 2D 46 61 41 A7 F6 DC AF D3 E6 00 00 1E
Packet size: 27
Sending response to 192.168.100.18:33339

[MiLightUdpServer port 5989] - Handling packet: 80 00 00 00 11 00 00 00 9B 00 31 00 00 08 04 01 00 00 00 01 00 3F Packet size: 2session: 0000, sequence: 155, group: 1, checksum: 63
Command cmdType: 31, cmdHeader: 00000804, cmdArg: 01000000
Sending packet (25 repeats): 
00 DB 47 79 66 D1 B4 66 54 
Patching existing state with: {"state":"ON"}
GroupState::patch: State changed: {"brightness":255,"bulb_mode":"white","color_temp":370,"color":{"r":255,"g":255,"b":255},"effect":"white_mode","kelvin":100,"level":100,"state":"ON","status":"ON"}
Elapsed: 219
Sending response to 192.168.100.18:33339
Flushing dirty state for 0xAABA / 1 / rgb_cct
[MiLightUdpServer port 5989] - Handling packet: 20 00 00 00 16 02 62 3A D5 ED A3 01 AE 08 2D 46 61 41 A7 F6 DC AF D3 E6 00 00 1E
Packet size: 27
Sending response to 192.168.100.18:33339

[MiLightUdpServer port 5989] - Handling packet: 80 00 00 00 11 00 01 00 9C 00 31 00 00 08 05 00 00 00 00 01 00 3F Packet size: 2session: 0001, sequence: 156, group: 1, checksum: 63
Command cmdType: 31, cmdHeader: 00000805, cmdArg: 00000000
Sending packet (25 repeats): 
00 DB 47 79 64 94 B5 66 06 
Patching existing state with: {"color_temp":370}
GroupState::patch: State not changed: {"brightness":255,"bulb_mode":"white","color_temp":370,"color":{"r":255,"g":255,"b":255},"effect":"white_mode","kelvin":100,"level":100,"state":"ON","status":"ON"}
Elapsed: 199
Sending response to 192.168.100.18:33339
[MiLightUdpServer port 5989] - Handling packet: 20 00 00 00 16 02 62 3A D5 ED A3 01 AE 08 2D 46 61 41 A7 F6 DC AF D3 E6 00 00 1E
Packet size: 27
Sending response to 192.168.100.18:33339

[MiLightUdpServer port 5989] - Handling packet: 80 00 00 00 11 00 02 00 9D 00 31 00 00 08 03 64 00 00 00 01 00 A1 Packet size: 2session: 0002, sequence: 157, group: 1, checksum: 161
Command cmdType: 31, cmdHeader: 00000803, cmdArg: 64000000
Sending packet (25 repeats): 
00 DB 47 79 61 5F C2 66 EB 
Patching existing state with: {"brightness":255}
GroupState::patch: State not changed: {"brightness":255,"bulb_mode":"white","color_temp":370,"color":{"r":255,"g":255,"b":255},"effect":"white_mode","kelvin":100,"level":100,"state":"ON","status":"ON"}
Elapsed: 198
Sending response to 192.168.100.18:33339
[MiLightUdpServer port 5989] - Handling packet: 20 00 00 00 16 02 62 3A D5 ED A3 01 AE 08 2D 46 61 41 A7 F6 DC AF D3 E6 00 00 1E
Packet size: 27
Sending response to 192.168.100.18:33339

[MiLightUdpServer port 5989] - Handling packet: 80 00 00 00 11 00 03 00 9E 00 31 00 00 08 04 01 00 00 00 01 00 3F Packet size: 2session: 0003, sequence: 158, group: 1, checksum: 63
Command cmdType: 31, cmdHeader: 00000804, cmdArg: 01000000
Sending packet (25 repeats): 
00 DB 47 79 66 D1 C3 66 5F 
Patching existing state with: {"state":"ON"}
GroupState::patch: State not changed: {"brightness":255,"bulb_mode":"white","color_temp":370,"color":{"r":255,"g":255,"b":255},"effect":"white_mode","kelvin":100,"level":100,"state":"ON","status":"ON"}
Elapsed: 195
Sending response to 192.168.100.18:33339
[MiLightUdpServer port 5989] - Handling packet: 20 00 00 00 16 02 62 3A D5 ED A3 01 AE 08 2D 46 61 41 A7 F6 DC AF D3 E6 00 00 1E
Packet size: 27
Sending response to 192.168.100.18:33339

[MiLightUdpServer port 5989] - Handling packet: 80 00 00 00 11 00 04 00 9F 00 31 00 00 08 03 64 00 00 00 01 00 A1 Packet size: 2session: 0004, sequence: 159, group: 1, checksum: 161
Command cmdType: 31, cmdHeader: 00000803, cmdArg: 64000000
Sending packet (25 repeats): 
00 DB 47 79 61 5F C0 66 E9 
Patching existing state with: {"brightness":255}
GroupState::patch: State not changed: {"brightness":255,"bulb_mode":"white","color_temp":370,"color":{"r":255,"g":255,"b":255},"effect":"white_mode","kelvin":100,"level":100,"state":"ON","status":"ON"}
Elapsed: 200
Sending response to 192.168.100.18:33339
[MiLightUdpServer port 5989] - Handling packet: 20 00 00 00 16 02 62 3A D5 ED A3 01 AE 08 2D 46 61 41 A7 F6 DC AF D3 E6 00 00 1E
Packet size: 27
Sending response to 192.168.100.18:33339

[MiLightUdpServer port 5989] - Handling packet: 80 00 00 00 11 00 05 00 A0 00 31 00 00 08 04 01 00 00 00 01 00 3F Packet size: 2session: 0005, sequence: 160, group: 1, checksum: 63
Command cmdType: 31, cmdHeader: 00000804, cmdArg: 01000000
Sending packet (25 repeats): 
[... 446 lines cut ...]
[MiLightUdpServer port 5989] - Handling packet: 20 00 00 00 16 02 62 3A D5 ED A3 01 AE 08 2D 46 61 41 A7 F6 DC AF D3 E6 00 00 1E
Packet size: 27
Sending response to 192.168.100.18:33339

[MiLightUdpServer port 5989] - Handling packet: 80 00 00 00 11 00 09 00 B1 00 31 00 00 08 03 05 00 00 00 01 00 42 Packet size: 2session: 0009, sequence: 177, group: 1, checksum: 66
Command cmdType: 31, cmdHeader: 00000803, cmdArg: 05000000
Sending packet (25 repeats): 
00 DB 47 79 61 3C BE 66 B8 
Patching existing state with: {"brightness":13}
GroupState::patch: State changed: {"brightness":13,"bulb_mode":"white","color_temp":370,"color":{"r":255,"g":255,"b":255},"effect":"white_mode","kelvin":100,"level":5,"state":"ON","status":"ON"}
Elapsed: 191
Sending response to 192.168.100.18:33339
[... cut ...]
[MiLightUdpServer port 5989] - Handling packet: 80 00 00 00 11 00 0A 00 B2 00 31 00 00 08 04 02 00 00 00 01 00 40 Packet size: 2session: 000A, sequence: 178, group: 1, checksum: 64
Command cmdType: 31, cmdHeader: 00000804, cmdArg: 02000000
Sending packet (25 repeats): 
00 DB 47 79 66 CA BF 66 48 
Patching existing state with: {"state":"OFF"}
GroupState::patch: State changed: {"brightness":13,"bulb_mode":"white","color_temp":370,"color":{"r":255,"g":255,"b":255},"effect":"white_mode","kelvin":100,"level":5,"state":"OFF","status":"OFF"}
Elapsed: 190
Sending response to 192.168.100.18:33339
Flushing dirty state for 0xAABA / 1 / rgb_cct`

sidoh, 2018-09-28:
thanks for providing that!

i see a crash in there, which is almost certainly what's going on.

i unfortunately won't have the chance to look at this for at least a couple of weeks.  but if anyone wants to poke around, this seems like a likely culprit: 

https://github.com/sidoh/esp8266_milight_hub/pull/335/commits/6eee8a951fa702061f6a2193aafcf6269502b7a1

poudenes, 2018-10-06:
Does someone had some time to look at this?

sidoh, 2018-10-08:
Thanks for looking, @amelchio.   I agree that it doesn't seem like the cause.

Probably missing a null check here:

https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/MiLight/RgbCctPacketFormatter.cpp#L50

poudenes, 2018-10-12:
Can someone build a new firmware so i can test it?

sidoh, 2018-10-20:
I'm able to replicate this using UDP, but not REST.  So issue is probably there.  Will hopefully have more time to debug today.

PaulGrz, 2018-10-24:
Well, just flashed 1.8.4 and it still crashes when trying to adjust brightness for group 0 (cct receiver)

sidoh, 2018-10-25:
Blah.  My editor ate one of the changes I made.  Just merged a fix -- will be in 1.8.5.

sidoh, 2018-10-25:
Yeah, the same bug existed for CCT bulbs.  It's fixed in 1.8.5.

https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.8.5