sidoh/esp8266_milight_hub #341: How to control with IObroker?

Mazdaracer121, 2018-09-22:
i can only control the first wifi bridge in iobroker (port5987)
Any other bridge i cant control even if i change the ports

sidoh, 2018-09-23:
I'm not familiar enough with iobroker to be helpful.  I'd be happy to answer more specific technical questions.

Mazdaracer121, 2018-09-23:
Umh Ok.. I think the Problem is that the esp have just one ip Adress... In milight app appears just the first bridge.

poudenes, 2018-09-23:
That is not helping. You see indeed only the first bridge. In milight app you cant change port

mbloos, 2018-09-26:
@poudenes : You can add multiple instances of the milight adapters (I had 3 before i switched milight to Home Assistant)
Then you will be able to change the port under the config.
And having the bridges on diffrent ports is possible (and with the same IP).

You would have multiple instances called milight.0, milight.1 and so on.

sidoh, 2018-10-21:
@Mazdaracer121 were you able to figure this out?  I'd be happy to add instructions to the "Integrations" section here:

https://github.com/sidoh/esp8266_milight_hub/wiki

Mazdaracer121, 2018-10-28:
Yes the bridge works now fine with my iobroker home automation Software. But when im slide the slider in saturation from 0 to 100, the slider jumps back to 0

Mazdaracer121, 2018-11-29:
Hey there. Is there any Chance to invert the Max and Min Values from Saturation? 0 to 100 and 100 to 0?

sidoh, 2018-12-02:
Can I ask why you'd want it to work that way?  The way it works now uses the correct definition of saturation.  100 saturation means the color is fully saturated, implying that the white LEDs are completely off.  And 0 saturation means the white LEDs are at full brightness.

sidoh, 2018-12-07:
Saturation = 100% should mean the white LEDs are completely off.   This is how my devices behave when I send espMH the command to update saturation to 100%.  Is this not the case for you?  If not, can you try controlling directly via the espMH UI or API?

Mazdaracer121, 2019-02-08:
Hey there. when im controlling from MilightHub Web interface, all works fine and right.
But i dont know why, Iobroker sends the Saturation Inverted...
Saturation 0, Leds on color, saturation 100, leds white

The easiest way for me, is invert the Saturation sent from Milighthub to the Bulbs...

Nahasapeemapetilon, 2019-02-19:
@Mazdaracer121 you could explain how you integrated it into iobroker? did you use the mqtt adapter ?

sidoh, 2019-05-19:
Sorry I dropped the ball on this one.

> But i dont know why, Iobroker sends the Saturation Inverted...
> Saturation 0, Leds on color, saturation 100, leds white

That's the incorrect way to interpret [saturation](https://en.wikipedia.org/wiki/Colorfulness#Saturation).  100 is 0% white, 100% color, 0 is 100% white, 0% color.

I'd be very happy to add instructions to the wiki if someone wants to take a stab, please add a comment to this issue.

Going to close, but please feel free to continue the discussion.

closed 2019-05-19 by sidoh