sidoh/esp8266_milight_hub #346: Not all lights in the same group are responding

jorisvanr, 2018-10-11:
Hi! I use this project in [DIYHue](https://github.com/mariusmotea/diyHue/) and in general it works quite ok. But 1 problem I have is that I have 6 GU10 lights in my ceiling that are all in the same group but are often not responding as 1 which can be really annoying. 

I added a gif to show what the problem is. It occurs both in the Hue App but also using the web interface of this project.

[image]

sidoh, 2018-10-11:
Think this has come up a few times, e.g., #296.

You can try turning up the number of repeats.  One comment in the issue linked above suggests interference with the ESP8266 and the nRF can cause problems.

sidoh, 2018-10-21:
Sounded like moving the nRF and ESP8266 further apart helped you, @jorisvanr?  Is it worth adding this to the [Troubleshooting Guide](https://github.com/sidoh/esp8266_milight_hub/wiki/Troubleshooting)?

sidoh, 2018-10-21:
Closing since it seems like this wasn't a software issue.  Feel free to yell at me if I'm wrong.

closed 2018-10-21 by sidoh

jorisvanr, 2018-10-21:
Yes! It helped me a lot! Thank you 🙏

sidoh, 2018-10-21:
added a section here -- https://github.com/sidoh/esp8266_milight_hub/wiki/Troubleshooting#interference

thanks for confirming!