sidoh/esp8266_milight_hub #348: New RGB+CCT Bulbs require unreasonable repeats!

iMartyn, 2018-10-19:
I do love this project but I'm running into fairly major problems now.

About half of my last batch (now ~3 years old) of RGB+CCT bulbs and all my current ones just won't work sensibly with the gw :(

Wemos D1 with multiple nRF modules - same behavior with old green 10-pin layout as newer black 8-pin fwiw.

I was messing about with repeats as some of them will just-about work sometimes, so I figure, get some more repeats.  Well, it works to an extent but to get guaranteed every-time colour-change I have to use in the region of 1800 repeats.

Unfortunately the esp just goes into meltdown if you do more than set a color or switch on/off with them that high.  Understandably as it takes about 6 seconds for the transmit to happen.

I unfortunately don't have an rgb+cct remote to test direct (I have some old original v5 gateways I can dig out but interestingly the default gw settings for this gw for RGBW also do not pair) but I have many esps and other micros that I can do tracing with if this helps, or potentially send you one of the RGB+CCT bulbs that is being "difficult".

I am suspecting that it's possibly that you've got the v6 protocol "close enough" but that it's not quite right and potentially with enough repeats, entropy is winning and the bulb is going "eh, close enough".  But this is just a guess as I know how hard reverse engineering can be.

Another "Interesting" thing I found and this may be intentional/an artefact of the protocol, but if I pair with one wemos and have the same settings on another wemos with the same radio and the same settings, nothing happens when I attempt to switch.  I was hoping to compare  and A/B test some things but that kinda gets in the way a little.

sidoh, 2018-10-20:
That does sound pretty unusable.

> I am suspecting that it's possibly that you've got the v6 protocol "close enough" but that it's not quite right and potentially with enough repeats, entropy is winning and the bulb is going "eh, close enough". But this is just a guess as I know how hard reverse engineering can be.

This is interesting, but there's no randomness in the packets that are sent, unless it's in some layer below that data that's sent to the nRF over SPI.  Given the same command, the same data is sent every time.

If I had to guess, you may have gotten a bad batch of nRFs.  Seriously, the build quality on those things is truly horrible [1].  In a batch of 10 I ordered, I have a few that are completely worthless, a couple that only kind of work (e.g., range problems, need lots of repeats, etc.), and a couple that work just fine.

How well does it work if you put the nRF really close to a bulb (also try different orientations)?

[1] https://forum.mysensors.org/topic/1153/we-are-mostly-using-fake-nrf24l01-s-but-worse-fakes-are-emerging/43

> Another "Interesting" thing I found and this may be intentional/an artefact of the protocol, but if I pair with one wemos and have the same settings on another wemos with the same radio and the same [...]

Think of "pairing" more as assigning bulbs a particular device ID to listen for, and less of assigning it to a particular gateway device.  You can just have them use the same device ID if you want to quickly test with different settings.

iMartyn, 2018-10-22:
Fantastic troubleshooting!  Indeed, some of my nRF modules are better than others too (and they're all >3yrs old).

So by isolating it down, I did indeed find that with extra power and trial and error, I have a setup at least on breadboard that works in the region of 20 retries reliably.

>Think of "pairing" more as assigning bulbs a particular device ID to listen for, and less of assigning it to a particular gateway device. You can just have them use the same device ID if you want to quickly test with different settings.

This is exactly what I believed to be the case but my experience wasn't matching up, but probably down to the radio not actually sending correctly was the real issue.

Apologies for thinking that the issue was at the software end.

closed 2018-10-22 by iMartyn