sidoh/esp8266_milight_hub #352: Cant load the web interface on desktop PC, can on phone.

richms, 2018-10-24:
This is a weird one I have here, 

2 issues and I think they are to do with the web UI, 

I just cannot get the web interface to load for me on my PC, I can ping OK, I can retrieve http://172.16.36.105/settings and http://172.16.36.105/about in the same browser just fine. Tried chrome, firefox, and edge.

I can see it on my phone OK, but it seems inoperative because the device ID field doesnt seem to want to let me choose anything in the dropdown that appears.

All do nothing on the PC. Edge will spin the wheel for a fraction of a second and then stop, chrome spin the circle in the tab forever and firefox has the reload button turn to the X like it is loading things, and nothing happens.

I have tried giving the device a new IP thru DHCP reservation, and same behaviour on that. Tried putting it on my SSID and vlan for trusted devices so it can see the internet, same again.

I can request http://itsIPaddress/whatever and get Notfound: /whatever as a response. Its just the actual root page to do anything that I cannot see anything on.

About:

{"version":"1.8.4","variant":"d1_mini","free_heap":24936,"arduino_version":"2_4_2","reset_reason":"Software/System restart"}

Settings:

{"admin_username":"","admin_password":"","ce_pin":16,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":50,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"","mqtt_username":"","mqtt_password":"","mqtt_topic_pattern":"","mqtt_update_topic_pattern":"","mqtt_state_topic_pattern":"","discovery_port":48899,"listen_repeats":3,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":200,"packet_repeat_minimum":3,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Slow blip","led_mode_packet":"Flicker","led_mode_packet_count":3,"device_ids":[8738,1638,57926],"group_state_fields":["state","color"]}

richms, 2018-10-24:
I think this may have been something in the flash from a previous use causing the problem with page loading. I wanted to get rid of the BS device IDs I had added since that was preventing me from using it on chrome on the mobile, so I flashed the wemos board from Arduino with the blink sketch and it set to overwrite all the flash, and then used nodemcu flasher to put the downloaded bin file on it again, and this time the UI loaded ok on the PC..

sidoh, 2018-10-26:
Cool, thanks for following up.  Clearing flash does indeed help sometimes :)

closed 2018-10-26 by sidoh