sidoh/esp8266_milight_hub #354: No saturation control with RGB+CCT bulbs

milobt, 2018-10-28:
I have an issue related to saturation of colour with FUT103 (RGB+CCT) bulbs.
I have two official MiHubs controlling around 20 bulbs around my house and then came across your project and loved the thought of having one hub to control all groups.

I have subsequently added some LED strip controllers (LS2).

I have your hub setup on a NodeMCU(v3) and works with HA, using MQTT Json however, I cannot saturate colours unless I pick a saturated colour from the colour wheel - is this normal with this version of bulb?
If the bulb is set to a colour using the HA colour picker and then I try to vary the saturation bar, the bulbs flicker white and then back to the colour.
The same happens with Alexa, if I set the bulbs to a colour and then try and set them back to white, they flick to white and then back to the colour.

Here's my HA light config:
- platform: mqtt_json name: "Kitchen Spots" 
state_topic: "milight/states/0xAD9/rgb_cct/1" 
command_topic: "milight/cmds/0xAD9/rgb_cct/1" 
brightness: true 
rgb: true 
color_temp: true 
effect: true 
effect_list: [white_mode, night_mode, colorfade_slow, colorfade_fast, 4] 
optimistic: false 
qos: 0 
retain: true

Are there particular 'Group state fields' that I should have selected in the Hub setup?

I also have another weird issue where not all bulbs in a group turn on or off at the same time, sometimes there is a delay or just don't come on/turn off.

Can you help?

sidoh, 2018-10-28:
For the saturation flapping, try disabling "Automatic Mode Switching" under Setup > Radio.

For the other issue, please work through the Troubleshooting guide:

https://github.com/sidoh/esp8266_milight_hub/wiki/Troubleshooting#hardware

Several users have reported improvements by increasing the physical distance between the nRF24L01 and ESP8266.

milobt, 2018-10-28:
Hey, thanks for the response.

Switching off auto mode switching worked for setting between colours and whites with Alexa.
In HA I can set colours in the colour wheel, including saturation, but if I try to change the saturation level - via the cool to warm white bar - it turns back to just white levels and colour is turned off.

Is this normal behavior for the setup now, using MQTT JSON light ?
I remember being able to to adjust the saturation levels with colour, with the normal MiLight Hub and HA light component?

I will experience with the distances between the PCBs and see if that improves performance, hopefully it will.
I have moved the hardware setup to different locations around the house to try and improve performance with mixed result.
For instance a bulb that was less than a metre away, would not consistently turn on/off, but others slightly further away would... :s

I am planning to design and 3D print an enclosure for the setup, but just need to take it to work and find time to design, but this will leave my home setup without its hub - the wife would not be pleased.

sidoh, 2018-10-28:
The cool->warm slider adjusts white temperature, not saturation.  It's expected that the bulbs switch back into white mode when adjusting white temperature.

I'm not aware of a way to adjust saturation direction within HASS.  You can adjust it directly with espMH in all of the usual ways, though - MQTT, REST, UI, UDP, etc.

Hopefully the troubleshooting guide helps with the problems you're describing.

Nice!  Curious to see what you come up with.  There's [this one](https://www.thingiverse.com/thing:2872591) on thingiverse, in case that helps at all.

sidoh, 2018-10-28:
Feel free to open another issue if you have further questions/problems that weren't addressed.

closed 2018-10-28 by sidoh