sidoh/esp8266_milight_hub #356: One Device ID group 4 switch on/aff all lights

cooltom, 2018-11-05:
Something really wird happend. I have 5 Device IDs and one buld from one group control all milights from all Device ID - basically all milights ... looks like it send some "universal" packet

I have:
0x15AB
0x25AB
0x35CD
0x45CD
0x55BB

if i send command on or off for 0x45CD in group 4 or ALL it switch on/off bulbs from all group all device IDs ...

packet looks like

ON

rgb_cct packet received (9 bytes):
Raw packet: 00 DB 1E E8 66 CC 30 61 9C 

Decoded:
Key      : 00
b1       : 20
ID       : 45CD
Command  : 01
Argument : 04
Sequence : 9A
Group    : 04
Checksum : 8D

OFF

rgb_cct packet received (9 bytes):
Raw packet: 00 DB 1E E8 66 D9 31 61 86 

Decoded:
Key      : 00
b1       : 20
ID       : 45CD
Command  : 01
Argument : 09
Sequence : 9B
Group    : 04
Checksum : 93

Funny is, taht it happend over the weekend. It was not happening before. I changed nothing. Now Tryed FW 183 and 185. Also unlink this bolb 4 from device ID 0x45CD and link it again. still same.

example of other packet from different group

rgb_cct packet received (9 bytes):
Raw packet: 00 DB CE 8A 66 D1 2E 66 E2 

Decoded:
Key      : 00
b1       : 20
ID       : 15AB
Command  : 01
Argument : 01
Sequence : 9C
Group    : 01
Checksum : 37

cooltom, 2018-11-08:
No it happend with another channel from another Device ID ... are the hex ID numbers to close each other? Is it possible?

sidoh, 2018-11-11:
I think I'm not following what your problem is.  It sounds like you have a bunch of bulbs paired to different device IDs, and bulbs paired to device ID `A` are responding to commands sent for some other device ID `B`?

Definitely shouldn't happen.

Can you provide a little more detail on:

1. What model of bulbs you're using?
2. Version of espMH (sounds like you're on 1.8.5?)
3. What protocol (sounds like rgb+cct?)
4. How many bulbs, which device IDs they're paired to, etc.
5. Which bulbs are responding to commands you don't expect them to respond to, which device IDs they're paired with, etc.

Keep in mind that newer bulbs allow you to pair with more than one device ID.

cooltom, 2018-11-11:
Huh, lot of investikation. So answers:

-it was not happening and just started over the weekend without any changes (i was not home and after i come back it started)

1. Using Milight GU10 RGB+CCT and GU10RGBCW and GU10 RGBWW -> all setup as RGB+CCT and i was able control them over ESP hub witout problems
the "universal command" is comming from 0x35CD channel 3 (+channel ALL) and 0x45CD channel 4 (+channel ALL), where GU10RGBCW are.

2. yes 1.8.5
3. yes  RGB+CCT 

4
0x15AB - 4x GU10 RGB+CCT
0x25AB - 4x GU10 RGB+CCT
0x35CD - 4x GU10RGBCW
0x45CD - 4x GU10RGBCW
0x55BB - 3x GU10 RGBWW

When I control 0x35CD - ch4/all GU10RGBCW - it control all bulbs from all Device ID (with or without paired bulb 0x35CD4 in socket)
When I control 0x45CD - ch3/all GU10RGBCW - it control all bulbs from all Device ID  (with or without paired bulb 0x45CD3 in socket)

So i guess its nt bolb related, no?

How to completly delete bulb from all device ID? I just umpair it from Device ID and pair back. But even if the bulb is umpared from Device ID channel and i send commands to this channel over ESPHub web interface its still happening.

I will wait if more test needed and than just unpair all 8 bulbs and delete 0x35CD and 0x45CD from ESPhub and setup 2 new Device IDs ...

Thnx

cooltom, 2018-11-27:
I changed Device ID 
0x35CD --->0x38CC
0x45CD --->0x48CC
if fix it. But after 2 days it start happening again :(

0x38CC2 is contorolling 0x48CC4, 0x48CC3, 0x48CC2, 0x48CC1
0x48CC3 is controlling 0x38CC1, 0x38CC2, 0x38CC3

It making me really mad. Seems like Milight RGBW have some problem ....

cooltom, 2018-12-06:
I changed Device ID
0x38CC --->0x30CC
0x48CC --->0x48CC
it fix it. But after 2 days it start happening again :(

**Seems like RGBW are useless with this hub :((((**

0x30CC3 is contorolling  all from 0x55BB
0x48CC3  is contorolling all from 0x55BB

Seems like aways different bulb and it always happend after few days.

Any idea???

cooltom, 2018-12-20:
It means group 3 -  0x48CC group 3
Wird is, that after i change ID and relink bulbs (unlink first, delete ID from hub and create new and link buld back) it works few days and than start happening again.
Its original Milights RGBW (Warm White) and RGBCW (Cold White)

sidoh, 2018-12-28:
Gotcha.

What I said in my last message is basically where I'm at: I need to be able to reproduce this in order to troubleshoot.

I own CCT bulbs that I test with, and have not noticed anything that even vaguely resembles this problem.

sidoh, 2018-12-31:
I use RGBW bulbs as well.

Some bulbs support being paired with several IDs.  When you re-assign IDs, are you unpairing bulbs with the old ID before pairing with a new one?

cooltom, 2019-02-08:
yes I removed it frst - buld light red. Than pair it with new ID

fab33, 2019-02-25:
I experience exactly the same problem with RGBW and RGBW+CCT. After few days, when sending commands to my RGBW bulbs (these are always powered), RGBW+CCT bulbs changed too... but RGBW+CCT are not always with power (I use hub only to change color) 
Since my bulbs are now always with power, no more problem. Crossing fingers ;)

sidoh, 2019-04-07:
I've not been able to reproduce this.  Please let me know if there's something I should try and I can reopen the issue.

closed 2019-04-07 by sidoh