sidoh/esp8266_milight_hub #357: LWT status messages

oakbrad, 2018-11-13:
Would it be possible to add Last-Will-and-Testament MQTT messages to the hub? For the purposes of tracking the connectivity of the device for problems. I think this is the only esp8266 project I use that doesn't have it.

https://www.hivemq.com/blog/mqtt-essentials-part-9-last-will-and-testament

sidoh, 2018-11-19:
I'm sure it's possible.  Happy to look at a PR.

sidoh, 2019-01-21:
This will be available in 1.9.0-dev5 (build is running now).

@oakbrad, please let me know if it does the trick for you.  I hardcoded willQoS to 2, but I can either change that or expose it as a setting if need be.

odis68, 2019-01-22:
Thank you so much!! made settings (MQTT LWT Topic) milight / states / LWT, MQTT LWT Message (Offline). Please make a place for the message also Online. Very convenient for processing in NODERED.

sidoh, 2019-01-22:
@odis68, I'm not following.  Are you asking for functionality where an MQTT message is sent when first connecting?

odis68, 2019-01-22:
If the device without power is offline. if you apply power Online.
when reconnecting to the hub comes the last state which leads to confusion.

sidoh, 2019-01-22:
Oh.  By "device," do you mean a light bulb or LED controller?

If so, then yes--the state constructed by espMH is "best effort."  There's no way for it to actually know what the state of a bulb is.  It only knows what commands it's sent and intercepted, and what the state resulting from those observations would be.

odis68, 2019-01-22:
I will enter the D1 mini controller with your code.

oakbrad, 2019-01-23:
@sidoh Thank you!! will try this out this weekend when I have some free time & report back.

subzero79, 2019-02-17:
I think is missing the part where you can set when the client comes online. For example i have sonoff touch controlling two gu10 milights.

If i know the esp8266 milight device is offline i will use the internal relay to control the light as failsafe. If the milight client becomes online (available) then i can use the milight hub to turn on/off the milights.

The zigbee2mqtt bridge implements the same for availability. So then in my automation software If it knows the software can't  send zigbee commands then it uses the internal relay toggle

In this case it should be hardcoded to the base path milight/ plus something like hub/state

In HA
```
- platform: mqtt
  name: "Living room"
  command_topic: milight/0x101/cct/1
  availability_topic: milight/hub/state
  payload_available: 'online'
  payload_not_available: 'offline'
  state_topic: milight/states/0x101/cct/1
  # <<: &MILIGHT_PARAMS
  schema: json
  color_temp: true
  brightness: true
  brightness_scale: 255
  retain: true
```

If the milight hub disconnects we should see something like this in HA

[image]

closed 2019-05-28 by sidoh