sidoh/esp8266_milight_hub #358: MQTT state after nightmode command

hencou, 2018-11-13:
Hi,

MQTT returns state=ON after a night_mode command.
Shouldn't it be state=OFF? Now, after a night_mode command, I have to swith off the lamp first in HomeAssistant before I can turn it on again.

sidoh, 2018-11-19:
Hmm... could probably go either way on this one.  Night mode seems more like an "on" state to me than an "off" state.

hencou, 2018-11-20:
Hi Sidoh,

Maybe I'm thinking in the wrong way in HomeAssistant?
I've created 2 components now to enable light and night mode now:
-A MQTT Switch:
  - platform: mqtt
    name: "Woonkamer nachtmodus" 
    command_topic: "mesh_in/milight/0x1225/rgb_cct/1"
    payload_on: '{"command":"night_mode"}'
    payload_off: '{"status":"off"}'
-A MQTT Light:
  - platform: mqtt_json
    command_topic: "mesh_in/milight/0x1225/rgb_cct/1"
    state_topic: "mesh_out/milight/0x1225/rgb_cct/1"
    brightness: true
    color_temp: true
    name: Woonkamer tuindeuren
    rgb: true

I agree with you that's more a On state then Off, but I don't know how to solve that on a more elegant way in HA. How would you do this?

sidoh, 2018-11-24:
I've seen some folks solve this with the `effect_list` feature of the mqtt_json component.

sidoh, 2018-11-24:
Not sure if that gives you exactly what you want, though.

amelchio, 2018-12-11:
Yes, using a night effect works fine for me.

For Home Assistant specifically, I also find it useful to enable [assumed state](https://www.home-assistant.io/docs/configuration/customizing-devices/#assumed_state). This gives separate on/off buttons like the MiLight remotes and not a toggle that can be wrong.

hencou, 2018-12-11:
@Amelchio: how did you set up that night effect in Home Assistant?

amelchio, 2018-12-11:
Like this: https://github.com/sidoh/esp8266_milight_hub/issues/319#issuecomment-418900096

hencou, 2018-12-12:
Thats a better approach! Works perfect here now. Many thanks.

sidoh, 2018-12-12:
Thanks @amelchio!

hencou, 2019-04-21:
The effect "night_mode" is not working in version 1.9.0 anymore.
Adding "state.fields._isSetBulbMode = 1;" to the function setNightMode in \lib\MiLightState\GroupState.cpp seems to solve the problem:

_```
bool GroupState::setNightMode(bool nightMode) {
  if (isSetNightMode() && isNightMode() == nightMode) {
    return false;
  }

  setDirty();
  state.fields._isSetNightMode = 1;
  state.fields._isNightMode = nightMode;

  **state.fields._isSetBulbMode = 1;**

  return true;
}
```_

sidoh, 2019-04-21:
Thanks, @hencou!  I've fixed this in #440.  Also added a test case to the integration test suite.

closed 2019-04-22 by sidoh

servim, 2019-04-22:
Hi! Which version will fix it?
Now filled esp8266_milight_hub_d1_mini-1.9.0-rc.4.bin
On two versions of the lamps works differently.
1. RGBW with night mode turned ON
RGBCCT when switching on the night mode the general switch does not change.
2. RGBW Lamps OFF.
Press - white, the lamps will turn on in the mode in which they were before shutdown, but the on / off button will remain OFF

sidoh, 2019-04-22:
1.9.0-rc.4 has the fix for the issue outlined by @hencou.

I believe the behavior you're describing has always been the case.  The RGBW packet formatter both turns on the bulbs and puts them into night mode.  The RGB+CCT does not -- it just sends the night mode command.

Not sure why, actually.  RGB+CCT bulbs will enter night mode regardless of whether they're off or on.  May not be the case with RGBW bulbs?

Other than the inconsistency being mildly confusing, does this cause you any trouble?

servim, 2019-04-22:
> Other than the inconsistency being mildly confusing, does this cause you any trouble?

I do not know yet, I did not think that they could behave differently.
What say on the second question?

> 2\. RGBW Lamps OFF.
>     Press - white, the lamps will turn on in the mode in which they were before shutdown, but the on / off button will remain OFF, then you can change colors and modes, but the button shows that the lamps are OFF.

sidoh, 2019-04-23:
Makes sense.  They might have to behave differently.  Milight's protocols are a bit of a mess. :)

Not entirely sure I'm following.  Are there some commands or screenshots you can share that reproduce the behavior?

servim, 2019-04-27:
[image]
1 line: lamps on in color mode
2 line: pressed off, the lights went out
3 line: pressed white, the lamps turned on, but the button - off
4, 5 line: changed color, lamps changed color, but button - off