sidoh/esp8266_milight_hub #363: Google Home change milight colour to orange instead of white

Kitser, 2018-12-19:
I use the Milight hub via Home Assistant for a couple of months now and it works great.
The only thing which is not working for me is to change the color from a rgbww bulb to white with Google Home / Assistant. When I ask my Google Home to change the color to white Google says she/he will change the color to white, but the bulb changes to orange. Changing to other colors via Google Home is working great.
I think it has something to do with the rgbww bulbs. Because I can change my rgb cct bulbs via Google Home to white as it should. 

When I set the color of rgbww bulbs to white via the Milight Hub web interface or via Home Assistant the bulb get white. So there is no problem.

My Home Assistant config looks like this

[code]
- name: 'Spot TV'
  command_topic: "milight/0x1111/rgbw/2"
  state_topic: "milight/states/0x1111/rgbw/2"
  <<: &MILIGHT_PARAMS
    platform: mqtt
    schema: json
    brightness: true
    rgb: true
    white_value: true
    color_temp: true
[/code]

When I change the last part to this I cannot change the rgb cct bulbs to white with Google Home.
[code]
    white_value: false
    color_temp: false]
[/code]
Then they behave the same as the rgbww. So I ask Google Home to change the color to white. Google Home says she/he will change the color to white. But the bulbs get orange.

I hope I don't have to replace all my rgbww bulbs for rgb cct bulbs in order to get this to work. Is there something I can do about it?

Kitser, 2018-12-22:
Maybe this helps to solve my problem.
I installed MQTT.fx and subscribed to "milight/states/0x1111/rgbw/2"
When I tell Google Home to change the color to white I see this message in MQTT.fx
`{"state":"ON","brightness":255,"bulb_mode":"color","color":{"r":255,"g":157,"b":0}}`

And when I set the color to white with Home Assistant I see this message:
`{"state":"ON","brightness":255,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}`

What can I do to let Google Home set the color to r":255,"g":255,"b":255?

sidoh, 2018-12-28:
I have no idea why Google Home is telling espMH to set the color to orange, but it sure looks like that's what it's doing. :)

How is Google Home integrating with espMH?

Try subscribing to the command topic (probably `milight/0x1111/rgbw/2` if you followed the README), as this is will show which commands are being sent to espMH.  The messages you're pasting are from the state topic.

Kitser, 2018-12-30:
To integrate espMH with Google Home I first added all my Milight bulbs in Home Assistant. And than add [the google_assistant component to Home Assistant](https://rc--home-assistant-docs.netlify.com/components/google_assistant/)

When I subscribe to milight/0x1111/rgbw/2 (command) and milight/states/0x1111/rgbw/2 (states) I can see what happens whenI tell Google Home to set a bulb to a specific color.

Blue:
Command: {"state": "ON", "color": {"r": 255, "g": 0, "b": 0}}
State: {"state":"ON","brightness":204,"bulb_mode":"color","color":{"r":255,"g":0,"b":0}}
My bulb changes to blue.

So far so good.

White:
Command: {"state": "ON", "color": {"r": 255, "g": 222, "b": 172}}
State: {"state":"ON","brightness":204,"bulb_mode":"color","color":{"r":255,"g":157,"b":0}}
At this point my bulbs changes to orange. It's good to know that this is a Milight rgbww bulb.

When I ask Google to change a Milight RGB CCT bulb to white this happens:
Command: {"state": "ON", "color": {"r": 255, "g": 222, "b": 172}}
State: {"state":"ON","brightness":204,"bulb_mode":"color","color":{"r":255,"g":222,"b":170}}
And this RGB CCT bulb changes to white.

So there is something wrong with the rgbww bulbs.
Maybe it is just not possible to change these bulbs to white via Google Home?

I did made an workarround:
In Nodered I made a sequence to change the payload from "r": 255, "g": 222, "b": 172" to "r": 255, "g": 255, "b": 255"
Now when I ask Google to change to white the bulb changes to orange and then changes to white (255,255,255) within a second. It's not perfect, but it works. 

If you have an other solution I'm happy to try it :-D

sidoh, 2018-12-30:
Thanks, that's helpful information.

Reading [the docs](https://developers.google.com/actions/smarthome/traits/colortemperature), I think this is probably expected behavior.  Google is told that the bulb doesn't support color temperature, so it's doing its best to create a warm white with RGB.

> Google may pick the appropriate mode to use based on request and light type (for example, Make the living room lights white might send Temperature commands to some bulbs and Spectrum commands to LED strips).

Perhaps you could add a white_mode effect and use [the Google Assistant "Scene" command instead](https://developers.google.com/actions/smarthome/traits/scene)?

You'd also need to update your light config to:

```yaml
- name: 'Spot TV'
  command_topic: "milight/0x1111/rgbw/2"
  state_topic: "milight/states/0x1111/rgbw/2"
  <<: &MILIGHT_PARAMS
    platform: mqtt
    schema: json
    brightness: true
    rgb: true
    white_value: true
    color_temp: false
    effect: true
    effect_list:
      - white_mode
      - night_mode
```

sidoh, 2019-04-07:
Going to close for now.  Please let me know if you'd like to discuss further and I can reopen.

closed 2019-04-07 by sidoh