sidoh/esp8266_milight_hub #364: Changed router errors

poudenes, 2018-12-22:
Hi All,

Used always a Apple TimeCapsule as router. Now have Linksys WRT 3200ACM

Since then i get errors in Domoticz:

2018-12-22 20:12:10.666  Error: AppLamp: Bridge not found, check IP Address/Port!...
2018-12-22 20:12:10.666  Error: AppLamp: Worker stopped!...
2018-12-22 20:12:10.666  Error: AppLamp: Bridge not found, check IP Address/Port!...
2018-12-22 20:12:10.666  Error: AppLamp: Worker stopped!...
2018-12-22 20:12:10.668  Error: AppLamp: Bridge not found, check IP Address/Port!...
2018-12-22 20:12:10.668  Error: AppLamp: Bridge not found, check IP Address/Port!...
2018-12-22 20:12:10.669  Error: AppLamp: Worker stopped!...
2018-12-22 20:12:10.667  Error: AppLamp: Bridge not found, check IP Address/Port!...
2018-12-22 20:12:10.670  Error: AppLamp: Worker stopped!...
2018-12-22 20:12:10.671  Error: AppLamp: Bridge not found, check IP Address/Port!...
2018-12-22 20:12:10.671  Error: AppLamp: Worker stopped!...
2018-12-22 20:12:10.671  Error: AppLamp: Bridge not found, check IP Address/Port!...
2018-12-22 20:12:10.671  Error: AppLamp: Worker stopped!...
2018-12-22 20:12:10.671  Error: AppLamp: Bridge not found, check IP Address/Port!...
2018-12-22 20:12:10.671  Error: AppLamp: Worker stopped!...
2018-12-22 20:12:10.669  Error: AppLamp: Worker stopped!...
2018-12-22 20:12:10.669  Error: AppLamp: Bridge not found, check IP Address/Port!...
2018-12-22 20:12:10.672  Error: AppLamp: Worker stopped!...
2018-12-22 20:12:10.669  Error: AppLamp: Bridge not found, check IP Address/Port!...
2018-12-22 20:12:10.673  Error: AppLamp: Worker stopped!...
2018-12-22 20:12:10.670  Error: AppLamp: Bridge not found, check IP Address/Port!...
2018-12-22 20:12:10.674  Error: AppLamp: Worker stopped!...

Can someone tell me what this can be?

Static IP for the module is still the same. Ports are also add in port foward

sidoh, 2018-12-22:
Is the IP still the same?  Can you use espMH manually without trouble?

poudenes, 2018-12-22:
Found problem. 
New Router was config with mixed wifi mode: 802.11.gn, changed it to 801.11g only
Since then when restart domoticz everything is working correct.

Only i have still a delay of actions... Already try changing some settings at radio but not help

sidoh, 2018-12-22:
Not exactly sure how to help with the delays, but glad you got it working.  Okay if we close this?

poudenes, 2018-12-28:
Yes you can close this.

closed 2018-12-28 by sidoh