sidoh/esp8266_milight_hub #366: Wifi connection

poudenes, 2018-12-30:
Hi,

Replaced my Apple Timecapsule for a Linksys WRT3200ACM with DD-WRT. I see some issues now. With Apple Timecapsule when i reset this one ESP disconnect wifi connection and when wifi is up again ESP connect again.

Now when i restart my Linksys the ESP still show the flashing light that its still connect, after the router is restart and running again. I cant access the ESP. I have to restart it and then it connect to Wifi and everything is ok.

Whan can this be? Because when my router is restart for some reason i always need restart the EPS manual also. When im from home by friend dont know how to do, so no lights.... hehe

sidoh, 2018-12-31:
I'm not sure why it would behave differently when you switched routers.  I use dd-wrt as well, and don't need to restart the ESP when I restart my router (I also have ~10 ESPs doing different things for which this is also true).

You could try reviewing serial logs while you restart your router.

sidoh, 2019-02-04:
@poudenes, please let me know if you continue to have issues.  Happy to look into it, but I'll need more information if you'd like help debugging.

closed 2019-02-04 by sidoh