sidoh/esp8266_milight_hub #369: Inconsistent pairing

slootjes, 2019-01-01:
After finally managing to get some kind of result I now run into issues with pairing:

- Bulb 1: paired OK and controlling it works as expected
- Bulb 2: paired OK and controlling it works as expected
- Bulb 3: paired OK at first (it flashed), could not control it

Refreshed the interface:
- Bulb 1 still works
- Bulb 2 and 3 are no longer present in the Device ID dropdown

I removed bulb 3 and tried pairing bulb 2 again; now bulb 1 started flashing and none of the bulbs respond to UI changes.

What is a good way to check what's happening?

sidoh, 2019-01-01:
Device IDs are not autosaved (see https://github.com/sidoh/esp8266_milight_hub/issues/282).

The right way to think about "pairing" is that you're associating your lighting device with a particular identifier, and not with a piece of hardware.  If you enter the ID you paired bulbs 2 and 3 with, it'll still work.

To save an ID after entering it, open the "Settings" dialog and hit save.

Most newer devices support being paired with more than one ID, which might explain some of the behavior you're seeing.

slootjes, 2019-01-01:
That makes sense, thanks! Maybe a good idea to also document this in the wiki? https://github.com/sidoh/esp8266_milight_hub/wiki/Pairing-new-bulbs

slootjes, 2019-01-01:
Maybe a stupid question but...the device ID, is that something I can just define myself or does the bulb broadcast this in some way? I'm unsure on what a good strategy is to set device IDs and also on how it relates to the Group ID. What I want to achieve eventually:

- 1 lamp in the living room with 5 bulbs
- 1 lamp in the kitchen with 3 bulbs

Both lamps should be controlled with their own switch and then control all bulbs at once. From what I understand I need to pair every bulb once at a time and then select group 1. For the second lamp I do the same thing but then select group 2. Is this correct?

sidoh, 2019-01-02:
Makes sense to me, I'll add it.  Thanks for the suggestion.

Yep, you can set it to whatever you want.  The wifi gateways you buy from milight have a random hardcoded device ID.  Each ID supports between 1 and 8 groups, depending on the protocol being used.  RGBW supports 4, FUT089 supports 8, etc.  espMH allows you to use whatever device ID you want per packet.

You can use separate device IDs per lamp, or use the same ID with different groups.  The behavior is equivalent.

sidoh, 2019-01-02:
Updated docs: https://github.com/sidoh/esp8266_milight_hub/wiki/Pairing-new-bulbs.  Let me know if you think anything is missing.

slootjes, 2019-01-03:
Nice! Sorry to be a pain but one thing I still don't see covered anywhere: how do the bulbs identify themselves? I assume they all have some kind of unique id which is stored somewhere. It would be great if this config is something that can be exported/imported to easily do pairing if someone is already aware of the bulb ids. Also it would allow to make changes to groups without unpairing and then pairing them all.

sidoh, 2019-01-03:
Except for some really weird corner cases (see https://github.com/sidoh/esp8266_milight_hub/issues/91), bulbs never send packets.  Pairing really just tells bulbs which Device ID to listen to.

I added a new section to the bottom of the [pairing page](https://github.com/sidoh/esp8266_milight_hub/wiki/Pairing-new-bulbs) explaining this.

sidoh, 2019-01-03:
(Realized I didn't respond to this part)

As far as I know, to make changes to the Device ID / Group ID that a bulb is paired with, it must be unpaired.

slootjes, 2019-01-03:
I think I now understand how it works however I can't get 5 bulbs to align on the same group. I pair them one by one but then a previously paired bulb starts to flash 6 times again as if it were new. Or when paired it loses it and then it gets into a state where it no longer listens for commands. Could it be something with the antenna or are these lights just not a reliable as I am expecting?

sidoh, 2019-01-03:
Some of the newer (last 3 years or so) devices support being paired with more than one Device ID.  It might do something silly like clear all previously paired IDs when it's maxed out.

When pairing them one-by-one, are the previously paired bulbs staying either on or off, or is the power being flipped on to them as well?  If so, they're probably responding to being paired as well.  Try either (A) pairing them all at once by powering them all on at the same time and hitting "Pair," or (B) pairing them one-by-one making sure previously paired devices are left off. 

When I pair new bulbs, I usually just screw them into a hot terminal and hit pair when they power on.

slootjes, 2019-01-03:
Good news! I switched the antenna from a regular NRF24L01+ to a NRF24L01+PA+LNA module I bought for $2 on AliExpress. I unpaired, then paired and all started flashing at once and now it seems to work just fine.

The silly thing is that I bought a new NRF24L01+ because the other one didn't work but it turns out this was because of the default pin 16 in the settings which should have been 4 (the PR you merged a few days back).

Thanks a lot for creating this app and providing this amazing support :)

closed 2019-01-03 by slootjes