sidoh/esp8266_milight_hub #374: Not able to send commands to bulbs

andrewmyhre, 2019-01-08:
I have been through the troubleshooting guide. 
I'm using a NodeMCU ESP8266 board made by HiLetGo.
I'm building and flashing the firmware myself using PlatformIO at commit b5e87b5d9c61207dc72919c6640e502eaaed9baf.
My bulbs are FUT104 bulbs and I'm using radio type RGB+CCT.

Output of http://milight-hub.local/settings:
```
{
"admin_username": "",
"admin_password": "",
"ce_pin": 16,
"csn_pin": 15,
"reset_pin": 0,
"led_pin": -2,
"radio_interface_type": "nRF24",
"packet_repeats": 50,
"http_repeat_factor": 1,
"auto_restart_period": 0,
"mqtt_server": "",
"mqtt_username": "",
"mqtt_password": "",
"mqtt_topic_pattern": "",
"mqtt_update_topic_pattern": "",
"mqtt_state_topic_pattern": "",
"discovery_port": 48899,
"listen_repeats": 3,
"state_flush_interval": 10000,
"mqtt_state_rate_limit": 500,
"packet_repeat_throttle_sensitivity": 0,
"packet_repeat_throttle_threshold": 200,
"packet_repeat_minimum": 3,
"enable_automatic_mode_switching": false,
"led_mode_wifi_config": "Fast toggle",
"led_mode_wifi_failed": "On",
"led_mode_operating": "Slow blip",
"led_mode_packet": "Flicker",
"led_mode_packet_count": 3,
"group_state_fields": [
"state",
"brightness",
"computed_color",
"mode",
"color_temp",
"bulb_mode"
]
}
```

Output of http://milight-hub.local/about:
```
{
"version": "1.8.5-4-gb5e87b5",
"variant": "huzzah",
"free_heap": 28440,
"arduino_version": "2_4_2",
"reset_reason": "External System"
}
```
I have 2 FUT104 bulbs and a MiLight Wifi hub.The lights do behave as expected when I use the app. 

I'm able to load the user interface and observe packets by turning on sniffing and sending commands to the bulbs using the MiLight app. The device ID is detected as 4CD4. When I enter `0x4CD4` into the device ID input I see a pop-out option `Add 0x4CD4...`, which I click to add it to the drop down list. When I reload the page the device ID drop down is empty however, so it doesn't seem to be persisting. It also isn't listed as a device in the settings json as you can see above. If I add the device and then click the controls there is no effect on the lights.

I have tried the above with 5 different NRF24L01+ modules from 3 different vendors. 1 of them is unable to detect anything during sniffing so I believe that to be defective (or just the wrong type of component). From what I've read even the fake NRF24L01+ modules are usually functional but just have different power draw profiles than the authentic modules, so I'm not convinced the problem is with the radio modules.

I'm a bit out of my depth working with electronics at this level so I'd appreciate some help with diagnosing the cause of the issue definitively so I can resolve it. The article about fake modules linked from the troubleshooting guide does mention some debugging techniques but they're quite difficult to parse for an Arduino novice.

Thanks!

andrewmyhre, 2019-01-09:
Thanks, I will try this. Regarding power supply at this stage I have both Arduinos running on USB from my laptop.

sidoh, 2019-01-12:
That almost certainly means that the radio modules you're using aren't working properly.  Hopefully the new modules do the trick.  I don't understand why these things are so hit-or-miss.

sidoh, 2019-02-04:
Please let me know if you're still having issues after trying different radio modules.

closed 2019-02-04 by sidoh

riker09, 2019-02-20:
@andrewmyhre I would be interested if the radio modules were the cause of the issue. I'm facing the same problems: I can sniff traffic, but I cannot send anything. When I send commands via the WebUI while sniffing I can see the packets. Not sure if they are just getting rendered or actually are picked by the antenna.