sidoh/esp8266_milight_hub #375: Set HSBColor command for use with new OpenHab 2.4 MQTT-Binding

DerOetzi, 2019-01-12:


sidoh, 2019-01-12:
Is there a reason to not use the colorRGB channel type instead?

sidoh, 2019-01-12:
Ah, I see.  I'd prefer to not have a bunch of platform-specific switching in espMH, but I'm open to it if there's enough demand.

Is it difficult to have some glue outside of espMH that maps to the appropriate values?

DerOetzi, 2019-01-12:
> Is it difficult to have some glue outside of espMH that maps to the appropriate values?

I have searched several days now before send this feature request to you and although analysed the source code of OpenHab/Eclipse SmartHome for this. For this I don't think there is a easy way to get another Output from the binding. I although looked at my other ESP-Firmwares which are Tasmota and ESPEasy both support this colorRGB format and Tasmota although colorHSB.

So I thought about another option which might be more suitable to implement, which would work for me. Maybe you can change your color command to except different formats as there are your current object, hexcode and my comma-seperated list like this:

`{color:{r:240,g:120,b:60} or {color:"#f0783c"} or {color:"240,120,60"}`

In my opinion this wouldn't be a platform-specific change, but an opening to support more different systems then just OpenHab 2.4.

So far thanks already for the really fast replies.

sidoh, 2019-01-12:
Gotcha.  Think it's a fair request, and wouldn't be tricky to add.  The only reason I hesitate is because each state field adds a little bit of overhead, and best to avoid that where possible.

sidoh, 2019-01-12:
Hi @DerOetzi, please try the 1.9.0-dev1 release:

https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.9.0-dev1

You'll find a new `oh_color` option under state fields.  De-select the other color fields, enable this one, and see if it works for you.

DerOetzi, 2019-01-13:
Hi,
sorry I'm very busy this weekend, will check this by Monday at the latest. And give you a feedback as soon possible. 

I fully understand the problem with the additional state field overhead. So what do you thing about my suggested solution to accept different input formats?

Thanks in advance

PheiPheiPhei, 2019-01-13:
Thank you so much for this!

DerOetzi, 2019-01-13:
Hi @sidoh,

I checked your changes, the part for state-output looks well. But my main goal was not achieved by your change. I need a input format to control the hub from openhab via MQTT.

Regards

sidoh, 2019-01-13:
Ah, of course.  This makes sense.  I'll add support.

> I fully understand the problem with the additional state field overhead. So what do you thing about my suggested solution to accept different input formats?

It's the same overhead :)

sidoh, 2019-01-13:
Please try [1.9.0-dev2](https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.9.0-dev2).

DerOetzi, 2019-01-13:
thanks again for your quick reaction. I really like projects with this kind of ownern. Unfortunately I want be able to check before monday evening, but will give you feedback as fast as possible.

From  my point of view as an software architect there is a difference of overhead and complexity between:
- There is a command to set color accepting rgb values in the following format: json object, hexcode, comma-seperated list or ...
- to set color in this format use command xyz, for set color in that format use command abc, for format 3 use cde...

sidoh, 2019-01-13:
I was never considering the second option. :)

sidoh, 2019-01-13:
Perhaps the misunderstanding is that it's a bit more complicated than just accepting commands in different formats because it's also necessary to have the ability to send state updates in different formats.  (This is where the `oh_color` state field comes in.)

Skinah, 2019-01-13:
Openhab can use RGB formats as I have proven this through...

A) MQTT rules which I now consider outdated. They can be found on the Openhab forum.
B) The newer binding which can be found here..

https://community.openhab.org/t/espmilighthub-new-binding-for-milight-limitlessled-and-easybulb/39720/

Internally Openhab does use the HSB (hue sat brightness), so it does take extra work to convert but it is possible to do this it just takes extra work on the programmers behalf. The binding has already done this and is simple to use.

DerOetzi, 2019-01-14:
> Perhaps the misunderstanding is that it's a bit more complicated than just accepting commands in different formats because it's also necessary to have the ability to send state updates in different formats. (This is where the `oh_color` state field comes in.)

Ah okay I see my missunderstanding. One option maybe would be to make for output a selection field or a string field where you can select or give a string-format for color output. But this is really difficult to decide whether this would be less overhead.

sidoh, 2019-01-14:
Yep, that's precisely part of the change that was made. :)

The functionality was already there for multiple types of fields; just a matter of implementing.

Anyway, this should be complete as of 1.9.0-dev2 (1.9.0-dev3 includes some additional, unrelated features).  Let me know if you run into issues.

DerOetzi, 2019-01-14:
It works like a charm. Although with 1.9.0-dev3.

Thanks a lot

sidoh, 2019-01-15:
Is this for group 0?

DerOetzi, 2019-01-15:
> Is this for group 0?

No for group 1 

`tele/milighthub/0x2/rgb_cct/1`

sidoh, 2019-01-15:
Can you paste the output of `/settings` ?

DerOetzi, 2019-01-15:
Here it is I only replaced MQTT username and password

`{  
   "admin_username":"",
   "admin_password":"",
   "ce_pin":4,
   "csn_pin":15,
   "reset_pin":0,
   "led_pin":-2,
   "radio_interface_type":"nRF24",
   "packet_repeats":50,
   "http_repeat_factor":1,
   "auto_restart_period":0,
   "mqtt_server":"10.0.10.3",
   "mqtt_username":"USERNAME",
   "mqtt_password":"PASSWORD",
   "mqtt_topic_pattern":"cmnd/milighthub/:device_id/:device_type/:group_id",
   "mqtt_update_topic_pattern":"",
   "mqtt_state_topic_pattern":"tele/milighthub/:device_id/:device_type/:group_id",
   "discovery_port":48899,
   "listen_repeats":3,
   "state_flush_interval":10000,
   "mqtt_state_rate_limit":500,
   "packet_repeat_throttle_sensitivity":0,
   "packet_repeat_throttle_threshold":200,
   "packet_repeat_minimum":3,
   "enable_automatic_mode_switching":false,
   "led_mode_wifi_config":"Fast toggle",
   "led_mode_wifi_failed":"On",
   "led_mode_operating":"Slow blip",
   "led_mode_packet":"Flicker",
   "led_mode_packet_count":3,
   "device_ids":[  
      1,
      2
   ],
   "group_state_fields":[  
      "state",
      "level",
      "oh_color"
   ]
}`

sidoh, 2019-01-15:
That all looks good.  This is indeed unexpected, although it's probably not related to this change.  Are you able to paste an MQTT log of this interaction?  Output of `mosquitto_sub` or something similar would be sufficient:

```
mosquitto_sub -h mqtt.server.com -p 1883 -t '#' -v -u user -P pass
```

DerOetzi, 2019-01-16:
Sorry I was already in bed when you wrote your request. Do you need another log from me?

sidoh, 2019-01-16:
Yes, a full log would be helpful.  I'm not able to reproduce.  We should also probably move this to another issue :)

DerOetzi, 2019-01-16:
I will record this today and send it as a new issue.

closed 2019-05-28 by sidoh

galinette2000, 2019-05-28:
Hi, what does this option do? I tried enabling it in MQTT settings, but i do not see any oh_color field in states topic.

It was reported that this option makes issues with the Domoticz plugin, but I can't reproduce since I don't see any effect on MQTT messages.

sidoh, 2019-05-28:
It uses the `color` key:

https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/MiLightState/GroupState.cpp#L830-L843