sidoh/esp8266_milight_hub #376: MQTT-State message incomplete

DerOetzi, 2019-01-16:
As reported the MQTT-State message on state topic doesn't report full state. Here a log where I grouped by actions I have done while recording.

``<initial>
	<MqttMessage id="1" timestamp="1547620472342" topic="tele/milighthub/0x1/rgb_cct/1" qos="0" retained="true" encoded="true">{"state":"OFF"}</MqttMessage>
</initial>
<turnon>
	<MqttMessage id="2" timestamp="1547620530097" topic="cmnd/milighthub/0x1/rgb_cct/1" qos="0" retained="false" encoded="true">{"color":"0,104,250"}</MqttMessage>
	<MqttMessage id="3" timestamp="1547620530152" topic="cmnd/milighthub/0x1/rgb_cct/1" qos="0" retained="false" encoded="true">{"state":"ON"}</MqttMessage>
	<MqttMessage id="4" timestamp="1547620530580" topic="tele/milighthub/0x1/rgb_cct/1" qos="0" retained="false" encoded="true">{"state":"ON"}</MqttMessage>
</turnon>
<colorchange>
	<MqttMessage id="5" timestamp="1547620560994" topic="cmnd/milighthub/0x1/rgb_cct/1" qos="0" retained="false" encoded="true">{"color":"0,79,250"}=</MqttMessage>
	<MqttMessage id="6" timestamp="1547620561362" topic="tele/milighthub/0x1/rgb_cct/1" qos="0" retained="false" encoded="true">{"state":"ON","level":100,"color":"0,76,255"}</MqttMessage>
</colorchange>
<turnoff>
	<MqttMessage id="7" timestamp="1547620615052" topic="cmnd/milighthub/0x1/rgb_cct/1" qos="0" retained="false" encoded="true">{"color":"0,0,0"}</MqttMessage>
	<MqttMessage id="8" timestamp="1547620615179" topic="cmnd/milighthub/0x1/rgb_cct/1" qos="0" retained="false" encoded="true">{"state":"OFF"}</MqttMessage>
	<MqttMessage id="9" timestamp="1547620615231" topic="tele/milighthub/0x1/rgb_cct/1" qos="0" retained="false" encoded="true">{"state":"ON"}</MqttMessage>
	<MqttMessage id="10" timestamp="1547620615696" topic="tele/milighthub/0x1/rgb_cct/1" qos="0" retained="false" encoded="true">{"state":"OFF"}</MqttMessage>
</turnoff>
<turnon>
	<MqttMessage id="11" timestamp="1547620625074" topic="cmnd/milighthub/0x1/rgb_cct/1" qos="0" retained="false" encoded="true">{"color":"0,54,250"}</MqttMessage>
	<MqttMessage id="12" timestamp="1547620625107" topic="cmnd/milighthub/0x1/rgb_cct/1" qos="0" retained="false" encoded="true">{"state":"ON"}</MqttMessage>
	<MqttMessage id="13" timestamp="1547620625296" topic="tele/milighthub/0x1/rgb_cct/1" qos="0" retained="false" encoded="true">{"state":"ON"}</MqttMessage>
</turnon>
<colorchange>
	<MqttMessage id="14" timestamp="1547620680699" topic="cmnd/milighthub/0x1/rgb_cct/1" qos="0" retained="false" encoded="true">{"color":"0,29,250"}</MqttMessage>
	<MqttMessage id="15" timestamp="1547620681066" topic="tele/milighthub/0x1/rgb_cct/1" qos="0" retained="false" encoded="true">{"state":"ON","level":100,"color":"0,29,255"}</MqttMessage>
</colorchange>
<turnoff>
	<MqttMessage id="16" timestamp="1547620701146" topic="cmnd/milighthub/0x1/rgb_cct/1" qos="0" retained="false" encoded="true">{"color":"0,0,0"}</MqttMessage>
	<MqttMessage id="17" timestamp="1547620701198" topic="cmnd/milighthub/0x1/rgb_cct/1" qos="0" retained="false" encoded="true">{"state":"OFF"}</MqttMessage>
	<MqttMessage id="18" timestamp="1547620701314" topic="tele/milighthub/0x1/rgb_cct/1" qos="0" retained="false" encoded="true">{"state":"ON"}</MqttMessage>
	<MqttMessage id="19" timestamp="1547620701820" topic="tele/milighthub/0x1/rgb_cct/1" qos="0" retained="false" encoded="true">{"state":"OFF"}</MqttMessage>
<turnoff>
``

sidoh, 2019-01-16:
The behavior you're seeing is expected, actually.  There are two things that'll probably make it obvious why it's the right thing:

(1) Bulbs ignore all commands other than being turned on while they're off, so computed state reflects that.  So for example, if you try to set the color while the bulb is off (or the hub thinks the bulb is off), it'll be ignored.

```xml
	<MqttMessage id="2" timestamp="1547620530097" topic="cmnd/milighthub/0x1/rgb_cct/1" qos="0" retained="false" encoded="true">{"color":"0,104,250"}</MqttMessage>
	<MqttMessage id="3" timestamp="1547620530152" topic="cmnd/milighthub/0x1/rgb_cct/1" qos="0" retained="false" encoded="true">{"state":"ON"}</MqttMessage>
	<MqttMessage id="4" timestamp="1547620530580" topic="tele/milighthub/0x1/rgb_cct/1" qos="0" retained="false" encoded="true">{"state":"ON"}</MqttMessage>
```

(2) By default, state updates are throttled to a rate of 500ms.  So if a command comes in when an update was sent out less than 500ms ago, internal state representation is updated, but an MQTT update is not sent out until 500ms have passed.  I believe that's what you're seeing here:

```xml
	<MqttMessage id="15" timestamp="1547620681066" topic="tele/milighthub/0x1/rgb_cct/1" qos="0" retained="false" encoded="true">{"state":"ON","level":100,"color":"0,29,255"}</MqttMessage>
	<MqttMessage id="16" timestamp="1547620701146" topic="cmnd/milighthub/0x1/rgb_cct/1" qos="0" retained="false" encoded="true">{"color":"0,0,0"}</MqttMessage>
	<MqttMessage id="17" timestamp="1547620701198" topic="cmnd/milighthub/0x1/rgb_cct/1" qos="0" retained="false" encoded="true">{"state":"OFF"}</MqttMessage>
	<MqttMessage id="18" timestamp="1547620701314" topic="tele/milighthub/0x1/rgb_cct/1" qos="0" retained="false" encoded="true">{"state":"ON"}</MqttMessage>
	<MqttMessage id="19" timestamp="1547620701820" topic="tele/milighthub/0x1/rgb_cct/1" qos="0" retained="false" encoded="true">{"state":"OFF"}</MqttMessage>
```

One thing I don't quite understand is that the last update should include `level` and `color` as well.  For example, this is what I see:

```json
milight/0x2226/rgb_cct/1 {"state":"OFF"}
milight/states/0x2226/rgb_cct/1 {"state":"OFF","brightness":128,"bulb_mode":"color","color":{"r":0,"g":255,"b":33}}
```

Could you request the state via the REST API?  Route for your gateway would be `/gateways/0x1/rgb_cct/1`.

DerOetzi, 2019-01-16:
I although send the sequences manually form my mqtt-client and checked the response of REST-API you have asked for. The response is exactly the same like on mqtt above even with more delay between each command because I had to type. So I'm not thinking anymore about a timin problem in this case

sidoh, 2019-01-16:
Are you able to compile the firmware with MQTT_DEBUG, DEBUG_PRINTF, and STATE_DEBUG turned on and review Serial logs?

DerOetzi, 2019-01-17:
I can try to do this the next days. Where are this constant be defined?

DerOetzi, 2019-01-17:
Is it okay for this debugging to use tag-version 1.9.0-dev3 or which version from git should I use?

sidoh, 2019-01-18:
Ah, in [`platformio.ini`](https://github.com/sidoh/esp8266_milight_hub/blob/master/platformio.ini#L29).

If you're not able to compile yourself, or if it'd be a pain to do so, I can upload a pre-compiled binary.

DerOetzi, 2019-01-18:
Thanks I found them I will setup an extra test system on my spare raspberry and spare d1 mini because on my live system there is too much traffic for a clear debugging.

odis68, 2019-01-19:
> Could you request the state via the REST API? Route for your gateway would be /gateways/0x1/rgb_cct/1

Is it possible to request the status via MQTT as it happens through GET (GET / settings), if so how to do it, if not, please add this function.

sidoh, 2019-01-19:
Yes, it's possible:

https://github.com/sidoh/esp8266_milight_hub#mqtt

There's no way to "query" MQTT for the status without changing the state, though.  Unless you count reading messages that are flagged as retained (state messages are).

DerOetzi, 2019-01-19:
Hi I didn't have the time to do the debugging. But I have found another hint for you howto reproduce the error. To switch of I first send command {"color":"0,0,0"} delay 600ms and after this send command {"state":"OFF"}. For test purposes I have removed the color 0,0,0 command and the delay and just send the state off, all states are send correctly. So it seems there is a problem with this color 0,0,0

odis68, 2019-01-21:
> There's no way to "query" MQTT for the status without changing the state, though. Unless you count reading messages that are flagged as retained (state messages are).

So I would like to receive the state without change for the work of the house at home.

sidoh, 2019-01-21:
@odis68, not currently possible, and I'm not convinced this feature makes sense.  States are published as retained messages, meaning new subscribes are sent the last known state when they connect.  If you'd like to continue discussion about this, please open a new issue.

odis68, 2019-01-26:
> @odis68, not currently possible, and I'm not convinced this feature makes sense. States are published as retained messages, meaning new subscribers are sent the last known state when they connect. If you'd like to continue discussion about this, please open a new issue.

Thanks, be sure to do it.

sidoh, 2019-04-07:
@odis68, I'm not able to reproduce this.  If you can find a reliable way to do it, please let me know and I can reopen the issue.

closed 2019-04-07 by sidoh