sidoh/esp8266_milight_hub #377: Device Identification/Support Needed: Possibly FUT005 no sniffing or data available

caffeineflo, 2019-01-18:
Hey,

I know this is not the support forum, yet, I couldn't find a better place and was hoping someone could help me figure this out so eventually I could bring back support for my issue to this project in a future PR.

I bought a lamp like the pic below a while back and figured given that the remote is 2.4GHz and the controller looks similar to some MiLight controllers that it might be milight and I could use this project to control my lamp. 
[image]
As it seems now, it's not quite milight. While the controller and remote responds to the same pairing/unpairing procedures, I can't sniff any data or pair it with my milight hub.

Initially I thought it might be my NRF24l01 chip or some wiring, but after testing 5 different NRF24l01 chips and several ESP's I'm certain it's not that.

Here are images of the controller and remote circuits. Initially I was thinking it's the FUT005 remote, but I'm uncertain about that now that it's not working.
[image]
[image]
[image]
[image]

If anyone could lead me in the right direction and/or help identify what I have here, that would be great! 

Thanks

sidoh, 2019-01-21:
I'm not sure this is a milight device.  Was it branded as such from the place you bought it?

Hard to tell without taking a closer look at the board on the remote.

caffeineflo, 2019-01-21:
@sidoh The picture of the blue PCB is the remote.

Reason why I believe it's a MiLight is that the exact same pairing and unpairing procedures work.

Any clue how I could further investigate?

sidoh, 2019-01-21:
What does the other end look like?  I'd recommend trying:

1. Check serial logs with DEBUG_PRINTF enabled
2. Sniff the SPI bus on the remote

caffeineflo, 2019-01-21:
Here are some more pictures or the remote. I'll crack out the buspirate and see if I can sniff anything on the SPI bus

[image]
[image]
[image]

sidoh, 2019-04-07:
I'm not sure this is a milight device.  Milight devices typically have a separate microcontroller and 2.4 GHz radio (LT8900).  That doesn't seem to be the case here.

Going to close for now, please let me know if you'd like to reopen.  I'd need a device to test with or SPI dumps to get much further.

closed 2019-04-07 by sidoh

caffeineflo, 2019-04-08:
As expected I haven't had time to do a SPI dump. Now I have both a logic analyzer and a bus pirate next to me, but where and how would you apply the logic analyzer to get an SPI dump as I can't identify the IC1 that would be the only thing I can think of here?!

sidoh, 2019-04-08:
Given there isn't a separate RF module (I don't see one anyway), maybe there's no SPI to sniff.

caffeineflo, 2019-04-09:
The remote is gone as of today. Seemed to had some error state and now I'm about to put in another receiver.

Thanks for your replies though!