sidoh/esp8266_milight_hub #378: Bulbs being missed as of 1.8.7 update

wwboynton, 2019-01-19:
Hey there!

I was previously using 1.8.5, having no issues. Today, I updated to 1.8.7 (stable) and the bridge promptly began missing groups when given several commands at once.

The commands are being invoked via Alexa, channeled through the [ha-bridge](https://github.com/bwssytems/ha-bridge) software and pointed at the esp board via the UDP feature.

When I downgraded by re-flashing the 1.8.7 firmware, the problem ceased. Happy to do more testing on request.

sidoh, 2019-01-19:
It's probably a crash.  Are you able to provide Serial logs?

sidoh, 2019-01-19:
Unfortunately, no way to get logs through the API.  Especially the kind where it's crashing. :)

Easiest way is to install PlatformIO and use the command `pio device monitor` with the ESP8266 attached to your computer via USB.  Depending on your OS, you might have to install some drivers.

I don't normally use UDP, and am not able to reproduce this with MQTT, so logs would be very helpful.

The changes from 1.8.5 -> 1.8.7 are pretty innocuous, but stranger things have happened :)

wwboynton, 2019-01-20:
I'm not precisely sure what you mean by "crash" in this case -- do you just mean erroring out? I'm used to "crash" meaning "go down," but the bridge definitely remains up and running. It just might only turn on three of the four groups I ask it to, and if I ask again it'll usually pick up the missing one on the second attempt.

In any case, I'll try to find some time to debug and pull some logs when I'm at home next!

sidoh, 2019-01-20:
When an ESP8266 crashes (watchdog timeout, memory error, etc.), it just restarts.  So missing commands or being momentarily unavailable is exactly the expected symptoms of a crash.

wwboynton, 2019-01-25:
Haven't had a chance to pull logs yet, but no longer observing this behaviour as of 1.9.0-dev6 :) will let you know if I see it again.

sidoh, 2019-01-25:
Huh, that's pretty strange.  No material changes in the 1.9 branch I can think of that'd affect this.  Please let me know if it happens again and I'll reopen the issue.

closed 2019-01-25 by sidoh