sidoh/esp8266_milight_hub #379: Triggering "all" group does not update state for individual groups

wwboynton, 2019-01-19:
Hey there!

I noticed in #284 you intended for "group 0" to adjust the state of the individual bulb groups within a virtual bridge.

As it goes, I'm not observing the correct behaviour. Sending an "OFF" command to group 0 via the web interface does not update the state of all four bulb groups in that virtual bridge, so issuing a toggle command to those individual bulbs takes two presses before the bulb actually changes.

I've observed this behaviour in 1.8.5 and 1.8.7 running on a NodeMCU ESP8266.

sidoh, 2019-01-19:
Ah, you're right.  I think I see why this is happening.  State for group 0 right now is just being flat-out ignored.  Working on 1.9 now, but I'll see if I can patch the 1.8 branch.

sidoh, 2019-01-20:
It was a little too messy to fix this in the 1.8 branch, so will be in the next 1.9 dev release.  Thanks for reporting!

galinette2000, 2019-05-19:
So will we have update of channels 1-8 when sending a command on group 0 ? Seems something required in order to implement the channel 0 (All) in Domoticz. Channels 1-8 states need to stay correct. Of course state of channel 0 has no real state, so we don't need any state update messages for channel 0.

closed 2019-05-28 by sidoh