sidoh/esp8266_milight_hub #383: Possibility to make the project running on Raspberry

lettore, 2019-01-23:
First of all thank you for this great project.
I think that everyone are using this system in some sort of ha system.
Usually such controllers are running on a Raspberry PI or a nuc, i think that add another device like the ESP8266 in the network will only add problems and risks of failure.
So the best option for me is to use the same Raspberry PI also for hosting the gateway software.
I'm using the MySensors Gateway on Raspberry that use the same radio you are using in this project, and also the zigbee2mqtt Hassio Addon for driving the Philips Hue Light system.
Another plus of making this available on Raspberry is to easily manage the service if something goes wrong and the best option for this is to make it available as a Docker container.
The Hassio addon is the best option because you can manage it from a web page within the Controller main page, but i know that not all are using or interested in using the Home Assistant controller.
Do you think will be possible in the future, or you know if someone is developing something similar?

sidoh, 2019-01-23:
This project is very targeted for the ESP8266 architecture.  It would be a huge overhaul to make it runnable on a Pi.

There are USB nRF24s, so certainly possible, but not something I'm currently planning on pursuing.  I'm not aware of others doing so either.

sidoh, 2019-01-23:
Some further discussion here: https://github.com/sidoh/esp8266_milight_hub/issues/58

closed 2019-01-23 by sidoh