sidoh/esp8266_milight_hub #384: LT8900 (long range) vs NRF24 (with antenna) for BEST STABLE PERFORMANCE

rawframe, 2019-01-24:
Hi Chris,
In your github page its says that ' is more performant.' If I buy a the long range 400m LT8900 (https://www.ebay.co.uk/itm/2-4G-150m-400m-Wireless-Transceiver-Module-Receiver-Transmitter-Super-24L01/332923935011?hash=item4d83d01923:m:m5cw9IqpeZRPN9mcmt3CPWw:rk:3:pf:0) will it be the best option for this project?
The github page also says 'NRF24s ... require software emulation' so how can i disable this if the LT8900 is being used instead of a long range (with antenna) NRF24L01 ?

Thanks in advance! Great work!!

sidoh, 2019-01-24:
I'd recommend using the nRF24 for now.

It's true that there's some overhead with the nRF24, but it's not really relevant for most use cases.  And there are far more nRF24 users, so I have more confidence that it works.

Only downside with LT8900 is that passive listens aren't supported.  Meaning it can't intercept commands from milight remotes, etc.

sidoh, 2019-01-24:
Please let me know if you'd like to reopen.

closed 2019-01-24 by sidoh

chmorant, 2021-01-04:
Hi Chris,

I'm more than tempted to replace my Milight boxes with your great hub.

But while doing tests in my office with 4 FUT066 at about 2 meters in clear line of sight I often get some of the lights to miss a change.
The Milight hub that is 7 meters away with two walls in the way extremely rarely get those misses.

I'm using a "Lolin" NodeMCU V3 with an nRF24 (I tried both antenna "long range" version and a standard one) with "1.10.7 (nodemcuv2)"

I increased the "HTTP repeat factor" to 4, played with the different radio settings but can't get the same stability.

I saw your reply regarding the LT8900 so I'm wondering what could be wrong with my setup.

Do you have any experience that could help me ? 

Thanks for your great work.

Christian