sidoh/esp8266_milight_hub #386: RGB_CCT should set bulb_mode to white on color_temp change

dlashua, 2019-01-25:
When an rgb_cct bulb is in bulb_mode: color, and a color_temp is sent, the bulb mode should change to white. Because it does not, setting color_temp when the bulb is in color mode has no effect and, in order to make the bulb white again one must first send an rgb_color of 255,255,255 and THEN send the desired color_temp. I cannot think of a situation where one would want to set the color temp without changing the bulb color from whatever color it is to the newly sent color_temp.

Without cached state, this would mean every color_temp change would require two commands to be sent to the bulb. Because milighthub has cached state now, if the rgb_color is already 255,255,255 that command doesn't need to be resent.

sidoh, 2019-01-25:
It should already behave in the way you've described.  Try disabling "Automatic mode switching" under "Radio."

sidoh, 2019-02-04:
Please let me know if this doesn't work for you.  Happy to reopen the issue if not.

closed 2019-02-04 by sidoh