sidoh/esp8266_milight_hub #387: NRF24L01+PA+LNA  (long distance) Would you recommend a 4.7uF 50v capacitor?

rawframe, 2019-01-26:
Hello, I have just bought **NRF24L01+PA+LNA** which is the long distance model with antenna. **Would you recommend a 4.7uF 50v capacitor between the ground and vin pins** for this device or should it work fine without? Thanks in advance

rawframe, 2019-01-30:
No one using NRF24L01+PA+LNA ?

garmck, 2019-01-30:
I am using a standard NRF24L01 but with an external antenna. I have followed this guide  https://www.instructables.com/id/NRF24L01-Fixing-Connection-Issues/ 
Adding a capacitor greatly improved reliability for me.
But I have seen these boards [Adapter-plate-Board](https://www.aliexpress.com/item/New-Socket-Adapter-plate-Board-for-8Pin-NRF24L01-Wireless-Transceive-module-51/32230227557.html?spm=2114.search0104.3.302.632279dax9Mgoc&ws_ab_test=searchweb0_0,searchweb201602_5_10065_10068_10547_319_317_10548_10696_453_10084_454_10083_433_10618_431_10307_537_536_10902_10059_10884_10887_321_322_10103,searchweb201603_45,ppcSwitch_0&algo_expid=937f4417-8a2c-4707-b312-f213dcc3aa80-46&algo_pvid=937f4417-8a2c-4707-b312-f213dcc3aa80 ) they take 5V and regulate to 3.3V

rawframe, 2019-01-31:
Thanks for the reply. I found this video which gives good instructions also
https://www.youtube.com/watch?v=bVpYhTBsOOE&vl=en

closed 2019-01-31 by rawframe

jshridha, 2019-02-16:
@rawframe Were you able to get your NRF24L01+PA+LNA to work? Mine seems to only want to receive but I can't get it to transmit.

rawframe, 2019-03-21:
> @rawframe Were you able to get your NRF24L01+PA+LNA to work? Mine seems to only want to receive but I can't get it to transmit.

Hi, my unit only just arrived. The **led flickers to indicate it is sending but I think nothing is being sent** since I cannot pair any bulbs. Did you have any success?

jshridha, 2019-03-21:
@rawframe Yes. I was able to get it to work. I had to put 2x 10uF capacitors across the 5V and ground pins. The problem is that the antenna draws more power than can be provided by the nodemcu. The capacitors act as a reserve of power that the antenna can draw from. Once I put those capacitors on, I was fine. You could also just put one 20 uF or 100 uF capacitor. I first tried with 1x 10uF and it worked about 70% of the time. After I added the second one, it's been working without issues.

rawframe, 2019-03-21:
@jshridha Thanks for the quick reply. I don't have any capacitors at the moment but I was able to connect a separate 3.3v supply to the NRF24L01+PA+LNA module, but this didn't work. I thought the NRF is a 3.3v device so when you say connect capacitors across 5v and ground pins are you referring to the esp or nrf module? cheers :)

jshridha, 2019-03-21:
You're right, it's a 3.3v device. I put it across power and ground on the NRF side.

rawframe, 2019-03-21:
thanks, given i have supplied 3.3v directly via a buck transformer I guess thevproblem is not only a lack of power but also smoothing ... Ill try a capacitor!

jshridha, 2019-03-25:
@rawframe I'm glad you got it to work. I wasn't able to find those settings. Can you point me to where they are?

rawframe, 2019-03-25:
@jshridha Power settings were exposed in the newer/latest build  under the radio tab in settings.  Because the interface contrast is low it can be confusing but the send channel user interface is not just a choice of low/med/high but rather you can select each option individually.