sidoh/esp8266_milight_hub #388: Colour temp sends 153 at both ends of the remotes slider. (rgb_cct)

Skinah, 2019-01-27:
If I slide my finger on the colour temp control of my remote and watch the MQTT states getting published I see a value of 153 at the cold white end, which is correct, then the numbers rise to 370 which is still correct, but if i slide further still I get 153 again at the wrong end of the slider. At the end where it should stop at 370 I get the wrong value. I suspect this is a simple bug and I have had it confirmed that another remote is also doing it as my first thought was it may be a bad remote.
Using latest 1.9.0 dev6 firmware.

Skinah, 2019-01-27:
Web GUI works correct it is only when you move a real remote.

sidoh, 2019-01-30:
Can confirm this happens on my remotes as well.  Bit of background on this:

* The raw values for color temp ranges in the RF packets are really wacky.  Documentation of the structure in [these comments](https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/MiLight/V2PacketFormatter.h#L32).
* Remotes send values outside of the range outlined in the "official" documents for the UDP protocol.
* There's already some builtin buffer room [here](https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/MiLight/V2PacketFormatter.h#L13), but looks like it's not enough.

I'll play around with increasing this buffer at some point for the 1.9 release.

sidoh, 2019-02-03:
Should be fixed as of 1.9.0-dev7 (build running now).  Thanks for flagging!

closed 2019-05-28 by sidoh