sidoh/esp8266_milight_hub #390: (MQTT) Support for FUT034

FrankMvanRijn, 2019-01-29:
Is it possible to add support for the FUT034 PIR?

sidoh, 2019-01-30:
Huh, wasn't aware that these things existed.  I ordered one, but will probably be many weeks before it gets here.

Does it show up in sniffed traffic at all?

FrankMvanRijn, 2019-01-30:
hi,

there is no traffic shown up in the sniffer when i activate the PIR, traffic from other remotes is shown up as usual. Let me know if i can do something to provide you information.

sidoh, 2019-01-30:
If you're able/willing, running with debugging enabled (DEBUG_PRINTF in platformio.ini is probably enough), would be interesting to see if anything shows up in serial logs.

Otherwise, I'll try to look into this when I get the device.

FrankMvanRijn, 2019-01-31:
Here the requested output:

Radio is available

Packet received: 900E569CF16D84309D6B0B3FPacket transformed: 09706A398FB6210CB9D6D0FCSuccessfully parsed packet of length 10
Received packet (len = 10)!
NRF24MiLightRadio - received packet!
NRF24MiLightRadio - Checking packet length (expecting 10, is 10)
Packet id: 28886
MiLightRemoteConfig::fromReceivedPacket: ERROR - tried to fetch remote config for unknown packet
WARNING: Couldn't find remote for received packet
Radio is available

Packet received: 90CD5C7B30EDA9300BPacket transformed: 09B33ADE0CB7950CD0Failed CRC: expected 20628, got 3280
Radio is available

Packet received: 903D78A4C245B067B5Packet transformed: 09BC1E2543A20DE6ADFailed CRC: expected 10491, got 59053
Radio is available

Packet received: 9081CC104E9C187FFCPacket transformed: 09813308723918FE3FFailed CRC: expected 42086, got 65087
Radio is available

Packet received: 903777ADC84CB929BC4BFC15Packet transformed: 09ECEEB513329D943DD23FA8Successfully parsed packet of length 10
Received packet (len = 10)!
NRF24MiLightRadio - received packet!
NRF24MiLightRadio - Checking packet length (expecting 10, is 10)
Packet id: 60626
MiLightRemoteConfig::fromReceivedPacket: ERROR - tried to fetch remote config for unknown packet
WARNING: Couldn't find remote for received packet

sidoh, 2019-02-04:
Looks like it's reusing the same packet format with a slightly different structure.  Decoded:

```
Packet : EC EE B5 13 32 9D 94 3D D2
Decoded: EC 2B 08 24 01 01 12 0C F1

Packet : 70 6A 39 8F B6 21 0C B9 D6
Decoded: 70 23 08 24 01 01 0E 0C F1
```

Can take a closer look when I get the device.  More dumps of this form would help:

```
Packet received: 903777ADC84CB929BC4BFC15
Packet transformed: 09ECEEB513329D943DD23FA8
Successfully parsed packet of length 10
Received packet (len = 10)!
NRF24MiLightRadio - received packet!
NRF24MiLightRadio - Checking packet length (expecting 10, is 10)
Packet id: 60626
MiLightRemoteConfig::fromReceivedPacket: ERROR - tried to fetch remote config for unknown packet
WARNING: Couldn't find remote for received packet
```

but it'll be easier when i have one to play with.

FrankMvanRijn, 2019-02-06:
Here some more output

Packet received: 906DF7CB666897F68BBBBF81Packet transformed: 09B6EFD36616E96FD1DDFD81Successfully parsed packet of length 10
Packet id: 46813

Packet received: 9060F9C168639928814F7C2EPacket transformed: 09069F8316C6991481F23E74Successfully parsed packet of length 10
Packet id: 1778

Packet received: 909DDF035C8007D3EF15832EPacket transformed: 09B9FBC03A01E0CBF7A8C174Successfully parsed packet of length 10
Packet id: 47528

Packet received: 9095D009528B0D59E3E082E6Packet transformed: 09A90B904AD1B09AC7074167Successfully parsed packet of length 10
Packet id: 43271

Packet received: 906CF6CA676996E78AAB6706Packet transformed: 09366F53E69669E751D5E660Successfully parsed packet of length 10
Packet id: 14037

Packet received: 9045C0FD5E4FA3ACADF31FDBPacket transformed: 09A203BF7AF2C535B5CFF8DBSuccessfully parsed packet of length 10
Packet id: 41679

Packet received: 903B7FA3C442B5E1B28FDBBFPacket transformed: 09DCFEC52342AD874DF1DBFDSuccessfully parsed packet of length 10
Packet id: 56561

Packet received: 90397CA0C641B773B13F6104Packet transformed: 099C3E056382EDCE8DFC8620Successfully parsed packet of length 10
Packet id: 40188

Packet received: 9064FEC46F669C438403140APacket transformed: 09267F23F66639C221C02850Successfully parsed packet of length 10
Packet id: 9920

Packet received: 9056DCE74D54BF11BB4353A3Packet transformed: 096A3BE7B22AFD88DDC2CAC5Successfully parsed packet of length 10
Packet id: 27330

Packet received: 90B2E62C65A622D8CE295CE1Packet transformed: 094D6734A665441B73943A87Successfully parsed packet of length 10
Packet id: 19860

Packet received: 90E86B4615D2943D3A676E75Packet transformed: 0917D662A84B29BC5CE676AESuccessfully parsed packet of length 10
Packet id: 6118

Packet received: 9078E1DC7F7682179C7D82EAPacket transformed: 091E873BFE6E41E839BE4157Successfully parsed packet of length 10
Packet id: 7870

Packet received: 905DD7EB4658B782B3CDC610Packet transformed: 09BAEBD7621AED41CDB36308Successfully parsed packet of length 10
Packet id: 47795

Packet received: 90114A8AEE7B92448B2FF5F3Packet transformed: 0988525177DE4922D1F4AFCFSuccessfully parsed packet of length 10
Packet id: 35060

Packet received: 90114A8AEE7B92448B2FF5F3Packet transformed: 0988525177DE4922D1F4AFCFSuccessfully parsed packet of length 10
Packet id: 35060

Packet received: 90CD5C7B30EDA99C0BEF2AFFPacket transformed: 09B33ADE0CB79539D0F754FFSuccessfully parsed packet of length 10
Packet id: 46071

Packet received: 909ADA0759840020E883E03DPacket transformed: 09595BE09A21000417C107BCSuccessfully parsed packet of length 10
Packet id: 22977

Packet received: 9048C9F15043A980A1D3BB01Packet transformed: 0912938F0AC2950185CBDD80Successfully parsed packet of length 10
Packet id: 4811

sidoh, 2019-02-23:
@FrankMvanRijn, I got the device.  Was sort of hard to get it working.  My feeling is that this is a unpolished and niche product that very few people are going to use (and that's relative to the rest of milight's stuff:).

The packet structure is different, but encoding is the same as other newer milight devices.  I can see patterns, but didn't spend much time digging in.

Honestly I will probably not spend time on this unless there are quite a few people who are interested.

Here are some notes on packet structure, in case anyone is interested:

```
[ 12 ] [ DB ] [ 08 24 ] [ 01 01 ] [ 34 ] [ 0C ] [ 35 ]
  ^      ^      ^         ^         ^      ^      ^
  |      |      |         |         |      |      |___ Checksum
  |      |      |         |         |      |__________ Probably brightness or night mode
  |      |      |         |         |_________________ Packet counter
  |      |      |         |___________________________ Always appear the same.  Probably brightness or night mode
  |      |      |_____________________________________ Remote ID
  |      |____________________________________________ (*) This is usually a constant remote ID.  Changes from packet to packet
  |___________________________________________________ Encoding key
```