sidoh/esp8266_milight_hub #393: Help with adding FUT096 like remote

halinka1125, 2019-02-06:
Hello
I have led strip RBGW with FUT096 like remote, how I can add support to this remote (I can`t sniff any data via web interface, but with debug_printf flag & nrf2401 I can see packets via putty console).
Bellow some output & putty log with all buttons 10x.
Thanks for help.
Packet received: E000ED0D0240 9C267DDF Packet transformed: 0700B7B040023964BEFB Successfully parsed packet of length 8
Received packet (len = 8)!
NRF24MiLightRadio - received packet!
NRF24MiLightRadio - Checking packet length (expecting 8, is 8)
Packet id: 100
Radio is available

Packet received: E000ED0D0240 EC267586 Packet transformed: 0700B7B040023764AE61 Successfully parsed packet of length 8
Packet received: E000ED0D0240 5C266B8B Packet transformed: 0700B7B040023A64D6D1 Successfully parsed packet of length 8
Packet received: E000ED0D0240 DC267013 Packet transformed: 0700B7B040023B640EC8 Successfully parsed packet of length 8
Packet received: E000ED0D0240 3C2660A1 Packet transformed: 0700B7B040023C640685 Successfully parsed packet of length 8
Packet received: E000ED0D0240 BC267B39 Packet transformed: 0700B7B040023D64DE9C Successfully parsed packet of length 8

[putty_2.log](https://github.com/sidoh/esp8266_milight_hub/files/2838141/putty_2.log)

sidoh, 2019-02-07:
FUT096 is supported, and those logs indicate to me that packets are being picked up just fine.  Are you able to open the developer console for the web UI?  Wonder if it's failing to connect to the websocket for some reason.

halinka1125, 2019-02-08:
Thanks for reply, webconsole is probably OK (If You think about about F12 console in Firefox - no errors indicated). I have just checket it - each device nodemcuv2+lt8920 and nodemcuv2+nrf2401 see packets sent by other device. But if I want to sniff my remote - web console is silent. Only nrf2401 device see packets (with debug flag enabled in fw) by serial port.So I think this is may be problem in protocol.

sidoh, 2019-02-10:
I have a FUT096.  It's working for me.  :\

sidoh, 2019-04-07:
Hi @halinka1125, I'm unable to reproduce.  Going to close, but please feel free to comment if you'd like to reopen.

closed 2019-04-07 by sidoh