sidoh/esp8266_milight_hub #395: Please add support for FUT037 RGB LED Strip Controller

Henry0191, 2019-02-10:
Hello,

could you please add support for the FUT037 RGB LED Strip Controller?
I can't pair with this type of controller.
Here's a link: https://www.amazon.de/dp/B01M0KU53O/ref=asc_df_B01M0KU53O58443770/

Thanks!

sidoh, 2019-02-11:
FUT037 works with several of the protocols supported by espMH -- 

http://futlight.com/productdetails.aspx?id=201&typeid=146

RGB+CCT, for example.

closed 2019-02-11 by sidoh

Henry0191, 2019-02-11:
@sidoh
No, it doesn't work. The wiring is correct, because I can sniff packages from my original milight hub. But pairing/unpairing/controlling devices from esp-mh doesn't work. I tried several times..

sidoh, 2019-02-11:
Please read through the [troubleshooting guide](https://github.com/sidoh/esp8266_milight_hub/wiki/Troubleshooting#bad-radio-module).

In particular, the radio module section:

> _Note that it's possible that you'll be able to receive, but not send packets._