sidoh/esp8266_milight_hub #396: MQTT disconnect hub all the time

poudenes, 2019-02-11:
Hi All,

tried everything. Domoticz is connected ok with username/password but the hub is disconnect all the time:

1549910697: New connection from 192.168.100.70 on port 1883.
1549910697: New client connected from 192.168.100.70 as milight-hub-7959806 (c1, k15, u'poudenes').
1549910720: Client milight-hub-7959806 disconnected.
1549910721: New connection from 192.168.100.70 on port 1883.
1549910721: New client connected from 192.168.100.70 as milight-hub-7959806 (c1, k15, u'poudenes').
1549910739: Client milight-hub-7959806 disconnected.
1549910739: New connection from 192.168.100.70 on port 1883.
1549910739: New client connected from 192.168.100.70 as milight-hub-7959806 (c1, k15, u'poudenes').
1549910867: Client milight-hub-7959806 disconnected.

Can someone assist me?

sidoh, 2019-02-11:
I'd suggest looking at Serial logs.

poudenes, 2019-02-12:
will do that later. Module is in a closet :D

Fjuxx, 2019-02-15:
just as an extra note, i also have these issues. so it does not seem to be an isolated case.

Running on a nodeMCU v2.

to bad it is connected to a power socket at the moment.
i will try to put it on a computer and get some serial logs from the devices when it happens again!

poudenes, 2019-02-15:
I changed something (read online)

milight/commands/:device_id/:device_type/:group_id

use this topic pattern and here it work now

Don't know if this is the real issue... but I don't have the problems anymore

Fjuxx, 2019-02-15:
that is the default topic. but i highly doubt that the topic name will result in complete MQTT disconnects.

Fjuxx, 2019-02-15:
Well i managed to break it down one way. but not sure if this is related to the random disconnects.

I spammed a lot of MQTT messages to the hub, and then it stopped working and in the serial logs i got:
`*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Already connected. Bailing out.
*WM: IP Address:
*WM: 192.168.1.89
Wifi connected succesfully

Setup complete (version 1.8.7)

Soft WDT reset

ctx: cont
sp: 3ffff920 end: 3fffffd0 offset: 01b0

>>>stack>>>
`
i skipped the stack as it seems like a memory dump only. let me know if you need that as well.

Is there perhaps a debug version that we can load for this?

sidoh, 2019-02-15:
Not having a command topic would probably cause MQTT to not work at all.  The library I'm using doesn't work well (maybe not at all) for publish-only.  Shouldn't matter what the topic is set to.

WDT reset means user code was spinning for too long without yielding to the system.

Yes, debug logs would be useful.  There are a bunch of flags you can enable in `platformio.ini` if you're able to compile an image yourself.  If you're not, let me know, and I can upload one.

Fjuxx, 2019-02-16:
oke running it now with some debug.

PS. i noticed that some packages were out of date, including the MQTT one. (first update in 3 years)
i did update everything, and am now logging the serial data to a file. will report back with results!

Fjuxx, 2019-02-16:
already had it failing once, but no real good logs around it:

did i miss a debug setting?
`MqttClient - publishing update to milight/states/0xDD90/rgbw/1
MqttClient - Got message on topic: milight/commands/0xDD90/rgbw/1
MqttClient - device DD90, group 1
Sending packet (50 repeats): 
B0 DD 90 00 01 04 21 

Soft WDT reset

ctx: cont 
sp: 3ffff930 end: 3fffffd0 offset: 01b0

>>>stack>>>`

This was a command that i send. didn't send anything for 15/20 min, and this was the first command again.
any idea's of what i can test?
or could this simply be a hardware issue?

rolfberkenbosch, 2019-02-20:
I have also the problem that mqtt disconnect. The webserver is reachable, so i go to settings and press the submit button at  mqtt settings and it is working again for 1 day. Maybe to build a keepalive for mqtt ?

sidoh, 2019-02-21:
WDT reset is always bad.  Can you post the full stacktrace, please? 

How is the nRF connected to the ESP8266?

Fjuxx, 2019-02-22:
I already made some changes to the source so i think the stacktrace wouldn't work for you would it then?

could you give me some info on how to decode the stack trace to see where it is coming from?
I tried adding a yield just after sending each packet, but yesterday the unit again crashed.

i did notice that i was still using D0 for the CE pin. so i swapped that over to D2 and changed the settings from number 16 to 4.

will see over the next couple of days if that works better.
but getting an idea on how to translate the stacktrace to actual code would be great!

sidoh, 2019-02-22:
Sure ---

1. Download/clone an exception decoder.  I use [this one](https://github.com/janLo/EspArduinoExceptionDecoder).
2. Paste the whole trace into a file.  I just use `/tmp/a`.
3. Run the decoder with something like:
  ```
  python /path/to/decoder/script/decoder.py -p ESP8266 -t ~/.platformio/packages/toolchain-xtensa -e .pioenvs/d1_mini/firmware.elf /tmp/a
  ```

notice the paths here.  The one after `-t` is the path to the xtensa toolshain, and is automatically downloaded by platformio.  Hopefully yours is in the same path.  After `-e` is the path to the `.elf` file, which should be generated along with the `.bin` when compiling the firmware.

> i did notice that i was still using D0 for the CE pin. so i swapped that over to D2 and changed the settings from number 16 to 4.

That shouldn't be a problem.  D0 just has some special behaviors during boot.

sidoh, 2019-02-22:
I've monitored my MQTT logs for several hours, and I don't see any disconnects.

It seems like there are two issues here:

1. If you're using MQTT, a command topic is required.  The MQTT library I'm using doesn't support not subscribing to at least one topic.  This has bitten me before.  If it's fixed in a future version, would be happy to upgrade.
2. WDT resets.  This is always bad; it's essentially a crash.  The whole system reboots.  Will need stacktraces to help debug.

The only common place I've seen WDT resets pop up is where wiring is loose.  Think what happens is there's a busy wait loop in the nRF library code that will trigger a WDT reset if the connection is faulty.  Try tightening the connection on the dupont wires, or -- ideally -- switch to soldered or pin-and-socket connections.

Fjuxx, 2019-02-23:
great thanks! i managed to decode it. (took a bit of tinkering with the script to get it to work on windows)

And indeed it seems to trace back to the sending to the RF24.
as i am currently using a breadboard to connect it i will replace it by direct wire's and see what will happen then!

this could also explain why it was so completely random.

thank you so far!

Fjuxx, 2019-03-11:
I can confirm that after replacing the breadboard with direct wire connection to the RF24 the problems have stopped!
(always hated analog stuff)

sidoh, 2019-04-01:
Great!  Gonna close.

closed 2019-04-01 by sidoh