sidoh/esp8266_milight_hub #398: Group 0 saturation remote movement causes empty JSON state.

Skinah, 2019-02-15:
Not sure if this is a bug?

Using a remote (rgb_cct only tested) with group 0 sends this MQTT message, which is strange as the bracket JSON is empty. All other functions work great with group 0, it is only the saturation that gives an empty json. Thank you for adding this recent group 0 state update feature. Using latest 1.9.0dev7 firmware

```
milight/states/0xE6C/rgb_cct/0 {}
```

sidoh, 2019-02-16:
State for group 0 is not accumulated. 

If you need to see what's going on for group 0, you can specify an MQTT "updates" topic, which will show individual commands.

sidoh, 2019-04-07:
This should actually be fixed in the latest dev branch.  I made some changes to the way group 0 is handled which should make behavior more intuitive.

closed 2019-05-28 by sidoh