sidoh/esp8266_milight_hub #400: Add button to control light

Mslawkowski, 2019-02-17:
I have an idea. next time you update, you can add options so that you can use gpio to connect a programmable button. it's possible to use the bell button instead of the ugly touch panel.

What do you think? 
written with a translator

sidoh, 2019-05-19:
Hi @Mslawkowski,

This one slipped through the cracks, I apologize.

It's a nice idea, although I'm a bit wary of adding too many features like this.  The software is already way more bloated than I expected it to be.

If many people want it, I'm happy to look into it.  But might be best off implementing it in a fork.

sidoh, 2019-05-27:
Gonna close this, but happy to discuss further if you'd like.

closed 2019-05-27 by sidoh

feder-ico, 2019-09-19:
Hi,
What if I want to ad a button to turn a digital pin on Hight or Low?

Should I include the button on index.html, but what else should I do to finally execute `digitalWrite(13, HIGH);` for example?

Any hint woul be appreciated!

Thanks in advance.