sidoh/esp8266_milight_hub #401: 1.9.0-dev7 Password not working

asking23, 2019-02-20:
Hi Version
1.9.0-dev7 is having some issue, when i tried NodemcuV2 .Bin file i was not able to connect using milightHub as password. 1.9.0-dev6 password is working fine. Please check this.

sidoh, 2019-02-20:
I see no changes in -dev6 -> -dev7 that could've possibly caused this.  This is the only commit: https://github.com/sidoh/esp8266_milight_hub/commit/3235b9e527f3e8cc49d761c389e19addde7fbd1e, which has nothing to do with wifi.

Maybe try again?  If it's still not working, revert back to -dev6 and make sure it works.  Also try clearing flash and reflashing dev7.

asking23, 2019-02-20:
I did try to empty nodemcuv2 and nothing...but works great with -dev6...

sidoh, 2019-02-22:
@asking23, I'm going to close this.  If you think there's something I missed in my effort to reproduce it, let me know.

closed 2019-02-22 by sidoh