sidoh/esp8266_milight_hub #402: Support for inwall remotes ?

Tontze, 2019-02-24:
Hi ! 

Been trying to get these two inwall remotest to work with this awesome bridge, i can see them sending commands, and can copy them to  be sent from home assistant, but they dont work .. 

Any advice where to go from here, or how to properly add them so they would work ? 

https://www.ebay.com/itm/Mi-light-2-4G-4-Zone-Touch-Screen-RGB-RGBW-Led-Controller-Wireless-RF-Remote-Con-/252693217228?var=&hash=item3ad5b07fcc

https://www.ebay.com/itm/Milight-Led-Panel-2-4G-wireless-4-Zone-RF-dimmer-Touch-Panel-rgb-led-Controller-/302353448837?var=&hash=item0

Tontze, 2019-02-24:
Lets say that i can control bulbs **directly** just fine, but when bulbs used by that remote, i dont get status updates to hass :/ I figured to change control from bulbs to that remote, i get status updates, but it wasnt so easy ..

garmck, 2019-02-27:
I'm using the B4 panel from your 2nd link. This works fine for me.
Are you using the same Device ID form the panel in HomeAssistant ?

Brignoud, 2019-02-27:
Me too, I'm using B4 also integrated in OpenHab with RGB CCT protocol
By the way curiously, I'm also managing CCT bulbs with it not only GU10 or other RGBCCT bulbs

Tontze, 2019-03-03:
I have B4  working now from HASS, but i dont get status updates to HASS when switching from B4.

Below is my setup for it : 

```
  - platform: mqtt
    name: 'Henna Light'
    state_topic: "milight/states/0x15BF/rgbw/1"
    command_topic: "milight/0x15BF/rgbw/1"
    <<: &MILIGHT_PARAMS
      # platform: mqtt
      schema: json
      color_temp: true
      rgb: true
      brightness: true
```

Tontze, 2019-03-03:
> I'm using the B4 panel from your 2nd link. This works fine for me.
> Are you using the same Device ID form the panel in HomeAssistant ?

What panel do you mean ? I picked Device ID from milight-hub sniffing

Tontze, 2019-03-08:
Should it even work thsi way ?

sidoh, 2019-03-08:
I think I'm not following what the problem is.  Given what you said here, it seems like you _are_ getting feedback when using the remote?

> This is what MQTT broker sees when operated ZONE1 from B4 switch :
>
> ```
>{"state":"OFF","brightness":245,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}
>{"state":"ON","brightness":245,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}
>```

long story short - yes, when espMH picks up a packet, a state update should be sent.

Tontze, 2019-03-09:
Propably i am getting the feedback, but why isnt HASS switch state updated ? But what you said here, problem must be on the HASS side ? Does my HASS config i provided earlier look ok to you ?

Tontze, 2019-03-11:
I made a switch with these settings (0x212 is hex from my remote): 

```
  - platform: mqtt
    name: 'Henna Light'
    state_topic: "milight/states/0x212/rgbw/1"
    command_topic: "milight/0x212/rgbw/1"
    <<: &MILIGHT_PARAMS
      schema: json
      rgb: true
      brightness: true
```
This is what mqtt shows when operating remote : 

```
{"state":"ON","brightness":235,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}
{"state":"OFF","brightness":235,"bulb_mode":"white","color":{"r":255,"g":255,"b":255}}
```
and from HASS: 

```
{"state": "ON"}
{"state":"ON","bulb_mode":"white","color":{"r":255,"g":255,"b":255}}
{"state": "OFF"}
{"state":"OFF","bulb_mode":"white","color":{"r":255,"g":255,"b":255}}
```
Same when sniffed with hub:

Remote: 
``` 
rgbw packet received (7 bytes):
Request type  : B0
Device ID     : 0530
b1            : 9A
b2            : 01
b3            : 04
Sequence Num. : 4F
rgbw packet received (7 bytes):
Request type  : B0
Device ID     : 0530
b1            : 9A
b2            : 01
b3            : 03
Sequence Num. : 4E
```
HASS: 

```
rgbw packet received (7 bytes):
Request type  : B0
Device ID     : 0212
b1            : 00
b2            : 01
b3            : 03
Sequence Num. : BA
rgbw packet received (7 bytes):
Request type  : B0
Device ID     : 0212
b1            : 00
b2            : 01
b3            : 04
Sequence Num. : B9
```

Should that mqtt be that different between sent from hass / remote ?

sidoh, 2019-03-16:
I think I'm not following all of the moving pieces you've got here.  It sounds like you have two remotes.  What are you trying to accomplish?  To have them both control the same light?

Have you tried setting something like this up?

https://github.com/sidoh/esp8266_milight_hub/wiki/Using-Milight-Remote-with-HomeAssistant

It forwards updates from one topic to another, which allows for this kind of behavior.

Tontze, 2019-03-16:
Well, i want my home assistant to be updated when turning lamp on/off from remote. I havent been able to config HASS to do that. I can successfully control all RGB, and such, from HASS and remote, but from remote HASS isnt updated .. 

My wall remote is paired with bulb. and i sniffed with milight hub remote ID and use that from hass. Is this wrong approach ? 

Thnx for the link, ill take a look at it.

Tontze, 2019-03-17:
link suggests setup that makes remote useless when HASS is shutdown so i couldnt control my bulbs at all ..

garmck, 2019-03-17:
You have your remote and hass on defferent device_ids
Change your hass config to match the real remote on 0x0530

Tontze, 2019-03-18:
ESPmh sniffed remotes id to 0530, and that should be changed to hex for hass setup, if google is right, it is 0x212 which i have in my hass setup. Is this correct ?

garmck, 2019-03-18:
Not sure why you need to convert to HEX?
Just use 0x0530 in HASS
I'm running multiple B4s and Hass is using the same sniifed Device ID as the B4 and its worked without issue. No HEX conversion used.

Tontze, 2019-03-18:
And your hass is updated when using B4? I converted To hex because docs said so?

garmck, 2019-03-18:
Yes. 
I’ve just re-read the docs and cannot see where it states you need to convert to hex.

Did you try using 0x0530 in your hass config yet?

Tontze, 2019-03-18:
Trying now. Heres the link : 

https://github.com/sidoh/esp8266_milight_hub/wiki/HomeAssistant#hass

Tontze, 2019-03-18:
OMG. it works :D It was all along that statement on the docs that it needs to be converted to hex :)

Tontze, 2019-03-18:
Closing this now.

closed 2019-03-18 by Tontze

garmck, 2019-03-18:
Great to hear.
You may want to remove **/:hex_device_id/** from your MQTT settings in the MQTT update topic pattern field.

Tontze, 2019-03-18:
Dont have anythin in it now, it was there just for a test :)