sidoh/esp8266_milight_hub #403: Once in a while my led bulbs are red and then i need to pair again

rolfberkenbosch, 2019-02-24:
Once in a while my led bulbs are red and then i need to pair again. It's comlete random on all mylights. Any idea? I use home assistant.

sidoh, 2019-02-25:
It sounds like they're somehow getting sent unpair commands.  Can you think of any way in which that might be happening?

Brignoud, 2019-03-03:
> It sounds like they're somehow getting sent unpair commands. Can you think of any way in which that might be happening?

Me too getting the same issue, as I wrote sometimes ago in your blog
The only thing I can report is a multi remote controller association and that this seems to show/happen when the normal wall switch is used

Skinah, 2019-03-15:
This usually happens when you send ON or OFF(?) commands directly after turning the power on for the globes. Why Milight choose such a common command to sync/unsync I have no idea as it should have been a separate code. Read the instructions on how to sync and unsync the globes, it states to use the power on/off buttons.

Brignoud, 2019-03-15:
I'm not sure I fully understand, sorry
You perheaps want to say that this happen when normally I switch directly from the wall?
Moreover, if it can help I desccribe the following:

1) Switch on the light from the wall
2) Governing from within Openhab OR Remote Controller
3) Switch Off from the Wall, not necessarily the same having multiple wall point (it's a wife effect! :-) )
4) Switch On from the Wall -> Bang, the light immediatly set to red color

Sometimes, directly change the colour from OH or Remote controller restore the situation
Sometimes, switching off/on affect one of the two of the same group and the red light doesn't respond until several attemps or changing from another remote controller
Thank you

sidoh, 2019-03-16:
Bulbs I've dealt with respond to being unpaired by turning red, blinking on and off a few times, and staying red afterwards.

The pairing/unpairing process involves:

1. Removing power from the device (e.g., switching the light socket off)
2. Supplying power back to the device
3. Within five seconds, sending the pair/unpair command

I'm not aware of another mechanism to unpair devices.

Either there's a way for bulbs to become unpaired beyond the above, or your bulbs are getting sent unpair commands being sent to your bulbs shortly after they're turned on.

@Brignoud is right -- the unpair command _is_ pretty dumb.  For most protocols, it's just the "On" command.

sidoh, 2019-04-07:
Going to close because I'm not sure there's much I can do about this.  Please let me know if you find a way to reproduce and I'll look into it.

closed 2019-04-07 by sidoh