sidoh/esp8266_milight_hub #404: Toggle fuction

agoralive, 2019-02-25:
Is there a way to create a "simple" Node-Red toggle button?

Perhaps a MQTT function which checks current state and then sends the opposite?

>  if state = on
>   then set state off
>  or if
>   stafe off
>   then set state on

The above is ABSOLUTELY now a correct code, but can anybody make one which works?

If the above is not possible could it perhaps be done by a MySQL db, where the latest state is stored in a DB which is then checked before choosing the state we wish for?

I would prefer if somebody could direct me to the DB solution, because I have hardware button which also is controlling the light, and would like to whole system to work together.

sidoh, 2019-02-25:
(assumed) state is stored internally and is accessible through the REST API and retained MQTT messages.  No need to keep track of it externally.

Where I have things that toggle on/off state, I just use HomeAssistant's toggle functionality.

Does this help?

agoralive, 2019-02-25:
No

I have a Xiaomi switch which controls the lights by converting the payload into a MQTT message, and whenever, a button in the NodeRed dashboard, the original panels or the remotes are turning lights on/off I would like to be able to change the state of a specific zone using a "button" in NodeRed, but this button does not know if the latest state is on or off.

I think I will try to find a solution where a db entry will set latest state, and then the NodeRed dashboard button will read this, react by choosing the opposite state, and then save the new state to the db for the next control and send an "update" MQTT for the ESP8266_milight_hub

sidoh, 2019-02-25:
There's no need to add another layer that keeps track of state.  espMH already tracks state.

agoralive, 2019-02-26:
Yes is keeps track but dosen't allow for "just" a state change. It is not possible to send a "toggle" command which changes the from "on to off" or "off to on".

Such an option would make it compatible with external button's AND Node-Red dashboard "button".

For instance the Sonoff custom firmware Tasmota, has such a function which allows for three (3) different devices to be controlled from one (1) Xiaomi wireless button (single, double and long press). This eliminating the need for a function, a db, and programming skills for the users. (Not that this is not good enough to be capable of, but not always an option)

garmck, 2019-02-26:
I think you need to raise an issue or look up details for NodeRed. 
I am doing exactly what you describe with a Xiaomi button, MiLight B4 panel and MiLight remote using MQTT to track states between all those devices. 
It seems you issue is a limitation of node red. Not espMH

agoralive, 2019-02-26:
@garmck Perfect

How do you update the MQTT statement for the bulbs, in order for the "button" in NodeRed Dashboard to switch off, in case the bulb is already on, by one of your hardware remotes?

garmck, 2019-02-27:
I'm not using NodeRed. As stated this maybe a limitation of NodeRed NOT espMH.

In espMH for MQTT settings have you configured topic, update topic and state topic ?

agoralive, 2019-02-27:
Yes everything works perfectly. There are no issues with, espMH, MQTT, Node-Red, Xiaomi, Sonoff's, etc

Only looking for a "toggle" function, which makes a button possible or a way to update a switch to actual state in Node-Red dashboard

agoralive, 2019-03-02:
If anybody has any idea's please share. I am looking for something like this:

 

> mosquitto_pub -t cmnd/device_id/POWER -m toggle

This command works with hardware which uses Tasmota software

sidoh, 2019-03-02:
It's not an unreasonable ask, although it'd look something like

```
mosquitto_pub -t milight/command/0x1/1 -m {"command":"toggle"}
```

sidoh, 2019-03-03:
Added in [1.9.0-dev8](https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.9.0-dev8).

agoralive, 2019-03-04:
@sidoh 

I have not tried this yet, but THANKS!!!!!!!!!!!!!!!!

closed 2019-05-28 by sidoh