sidoh/esp8266_milight_hub #405: ESP32 support

zyrorl, 2019-02-27:
Is there any chance of a version of the software targetting ESP32 builds?

sidoh, 2019-03-03:
Probably wouldn't be too hard, but taking advantage of the (dual core) hardware would be a lot more challenging.  My guess is that there are a chunk of ESP8266-specific things that would need to be wrapped in conditional macros, with ESP32-specific replacements added.

maxdd, 2019-03-14:
I'm also super interested in the ESP32 porting!!

jellewie, 2020-01-07:
I personally like the ESP32 more, only reason I goe ESP8266 was for this project, I have like 10+ ESP32 at home.
So I'm also interested in an ESP32 port!

ferventgeek, 2020-01-20:
Also would like to see an ESP32 version. It's SPI to the 2.4 breakout and the code looks to be adaptable for the REST WiFi. Anyone else working on it? I'm happy to help test.

mrtncls, 2020-08-15:
I ordered some ESP8266s to run MH but have some spare ESP32s laying around.  
I spend a few hours on getting it running on ESP32 and it works. No all features are working but the core is :-)

See my ESP32 branch: https://github.com/mrtncls/esp8266_milight_hub/commits/esp32

sidoh, 2020-08-15:
Beautiful! Thanks very much for taking a crack at this, @mrtncls.

Wanna toss it into a PR?

mrtncls, 2020-09-03:
I had the ESP32 running for about a week.

In the beginning, switching light from the UI (webinterface) and MQTT worked smooth. but after running more then a day, the MQTT events were processed slower and slower. In my MQTT client, I saw the update comming in fast but the ESP32 reacted only after 5-10 seconds. And when up even longer, it didn't process at all. It feels like a memory issue but I didn't spent more time to investigate.

In meantime, my ESP8266 arrived. It seems to work more stable. Didn't had any slowness issues until now (running a almost a week).

I only have one antenna. Once I have another one and can find some time, I'll dig into this.
I'm glad to help if someone else want to have a look.

butteryak, 2020-10-18:
Curios if any more work or progress was made on this port? tnx+  would love to have an ESP32 option++  I'd love to have an option to have an ethernet port such as this: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-ethernet-kit.html

jellewie, 2020-10-19:
@butteryak I did make a port (/new) ESP32 version of the [switch](https://github.com/jellewie/Arduino-Smart-light), but I am unable to port this due to the complexity and that's in platformIO. For now, this is the only 8266 that I own, For now, I'm just patiently waiting on the day there is ESP32 support

mrtncls, 2020-12-19:
When testing the ESP32 build, I mentioned that I had issues.
I had similar issues with my ESP8266 setup.

I figured out that my wifi router was the root cause. Milight works on 2.4Ghz like wifi. I guess the beamforming feature of my router caused network issues from time to time on my milight hub.
After switching the router's wifi to legacy mode, it works stable.

During my testing, I had the ESP32 running in parallel and it was stable.
I'm now running for a week on the ESP32 without issues.

butteryak, 2020-12-19:
What ESP32 are you using? I'm interested in trying this.  Particularly, I'm hoping to use a wired build such as this, possible you think?  https://wesp32.com/

mrtncls, 2020-12-21:
I'm using a Wemos Mini D1: https://nl.aliexpress.com/item/32916274028.html

You can use any PlatformIO supported board by setting it in the platformio.ini file.

I've also looked into LAN support because I had wifi issues.
But for this, we need to change code to have support for those LAN boards.
The LAN module is a seperate module and is outside of the ESP32 SOC/framework. Firmwares like ESPHome and Tasmota support this already.

butteryak, 2020-12-21:
That makes sense, I remember reading somewhere that the base of the milight hub code was tied into the wifi code on the 8266.

Certainly adjusting platformio.ini to meet and building firmware to meet the needs of my board is not beyond my abilities, but integrating the LAN module is certainly beyond my paygrade ;-) 

Any thoughts or plans to try?   I at the very least could help troubleshoot a bit.