sidoh/esp8266_milight_hub #406: nodemcu china clone - which pre-compiled binary to flash? can't access web-ui...

coolduke666, 2019-03-02:
hi

I have two questions:

a) which binary do you recommend for nodemcu china clone with a ESP8266EX with 4MB flash? I looked in the "platformio.ini" there not much difference between your precompiled binary for the nodemcu and d1-mini. so why do you provide different files?

b) when I flash the nodemcu file (latest 1.9.0-dev7), the ESP reboots and I can access the "ESPxxxx" AP and configure my wifi, then it reboots and nothing :( It doesn't receive any IP from my DHCP server, hence I can't access the web-UI. I also tried the D1-mini binary with the same result. Any idea how to proceed?

thx for your support!

sidoh, 2019-03-03:
Hello,

> a) which binary do you recommend for nodemcu china clone with a ESP8266EX with 4MB flash? I looked in the "platformio.ini" there not much difference between your precompiled binary for the nodemcu and d1-mini. so why do you provide different files?

Totally valid question.  There isn't a difference between the `d1_mini` and `nodemcuv2` platforms that I'm aware of.  To be honest, I added both because they seemed distinct at the time, even though they're not.  And removing them would probably just create more questions than I'd like to deal with.

> b) when I flash the nodemcu file (latest 1.9.0-dev7), the ESP reboots and I can access the "ESPxxxx" AP and configure my wifi, then it reboots and nothing :( It doesn't receive any IP from my DHCP [...]

I'd check the serial logs to see what's going on.

sidoh, 2019-03-03:
( oh -- and either binary should work.  as far as i'm aware, the only real difference in the different ESP8266 platforms I'm compiling against is flash size, and they're almost all 4MB. )

coolduke666, 2019-03-03:
thx.. do I have recomplie it in order to get serial logs or are the enabled in the default binaries?

access serial via usb or via rx/tx pins on the ESP? 

thx

sidoh, 2019-03-03:
You can recompile to enable some noisier debug logging, but default logs should be enough.

I've always accessed over USB.  Most dev boards have a USB to UART converter builtin.  I access them with platform io using the command `pio device monitor`.

coolduke666, 2019-03-06:
hi,

I managed to enable the console - here are the logs. Any idea?

after flashing:

*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 0
*WM: 
*WM: Configuring access point... 
*WM: ESP1344214
*WM: milightHub
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started

... config Wifi

*WM: WiFi save
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Connection result: 
*WM: 3
Wifi connected succesfully

Setup complete (version 1.9.0-dev8)

--> reset via button

                                  pX�CGH�@:h�0��*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.1.228
Wifi connected succesfully

Setup complete (version 1.9.0-dev8)

BUT: I can't access 192.168.1.228

browser:

This site can’t be reached http://192.168.1.228/ is unreachable.

ping 192.168.1.228
PING 192.168.1.228 (192.168.1.228) 56(84) bytes of data.
From 192.168.1.167 icmp_seq=1 Destination Host Unreachable
From 192.168.1.167 icmp_seq=2 Destination Host Unreachable
From 192.168.1.167 icmp_seq=3 Destination Host Unreachable
^C
--- 192.168.1.228 ping statistics ---
5 packets transmitted, 0 received, +3 errors, 100% packet loss, time 69ms

sidoh, 2019-03-07:
Looks completely normal.  Is it possible your router is blocking traffic somehow?

coolduke666, 2019-03-07:
can't be because they are on the same switch. that's really weird :(

Tontze, 2019-03-08:
Can you see it from your router that its really connected ?

sidoh, 2019-03-16:
I'm not sure what to tell you.  Looks like the esp8266 thinks it's connected.

I would suggest flashing it with different firmware.  If you still have connection issues, obviously the issue has something to do with either your esp8266 or your router.

I would expect the esp8266 to be complaining in serial logs if there were troubles, so i suspect something else is going on.

I'm going to close the issue because it doesn't seem like it's a problem with espMH.  Happy to keep discussing, or to re-open if you find evidence to the contrary.

closed 2019-03-16 by sidoh

agoralive, 2019-05-06:
@coolduke666 - Do you have a link for the esp you are using? I have the same problem with this model: https://www.aliexpress.com/item/1pcs-NodeMCU-V3-Lua-WIFI-module-integration-of-ESP8266-extra-memory-32M-Flash-USB-serial-CH340G/32813084968.html?spm=a2g0s.9042311.0.0.27424c4dZslw1E

mbd-shift, 2020-10-07:
I also use a clone which is branded ESP8266MOD, has 2x8+6 pins and is on a board that says "D1 mini".

I have the same issues.
Boot up works until the password is set.
Depending on which binary I used (tried all of them), there happens one or the other:
1. It connects to the router where it is listed under connected WiFi devices, then immediately disconnects after 5 seconds
2. It connects, is visible under Devices and has an IP address, but cannot be pinged or accessed. Host unreachable.

i used esptool.py for the upload, also tried with lower baud rate and different flash modes.
All the same, no connection.

It seems the device freezes after I get the serial message ``*WM: Connecting as wifi client...``
it doesn't even display the IP in serial, just drops "dead".

Chip Identifies as ESP8266EX and with a 26mhz crystal and 4MB Flash Manufracturer "5e"