sidoh/esp8266_milight_hub #408: china RGB milight clones

coolduke666, 2019-03-07:
hi

does anybody knows if the cheap china milight clones for RGB led stripes (controller plus remote approx 7 USD) works with this hub? or are the using another proprietary protocol which is different? 

remote looks like FUT098 ;) 

thx

sidoh, 2019-03-07:
I personally have no idea :)

coolduke666, 2019-03-10:
to all: china clones are using a non compatible protocol and therefore are NOT usable with the milight hub :(

sidoh, 2019-03-16:
Thanks @coolduke666.  Happy to dig into this a little more if you have more information.

closed 2019-03-16 by sidoh

afska, 2023-10-02:
any news on support for this controller?

@coolduke666 does your rgb controller look like this?

[image]