sidoh/esp8266_milight_hub #411: Lamp FUT-062/Controller FUT-089 Not Pairing

marcogastali2018, 2019-03-10:
I can not pair at all and I've tried everything. Lampada FUT-062 please, what to do. Follow log:

fut089 packet received (9 bytes):
Raw packet: 00 D8 A6 41 66 CE BC 63 0A 

Decoded:
Key      : 00
b1       : 25
ID       : CD62
Command  : 01
Argument : 02
Sequence : 0E
Group    : 02
Checksum : 1F

sidoh, 2019-03-10:
Are you sure packets are being received by the bulb at all?

Please read through the [troubleshooting guide](https://github.com/sidoh/esp8266_milight_hub/wiki/Troubleshooting).

marcogastali2018, 2019-03-10:
Packages are not being received by the lamp. I have no sign on the lamp.

sidoh, 2019-03-10:
Right.  Please read through the aforementioned troubleshooting guide.  You probably have a defective nRF24 module.

marcogastali2018, 2019-03-10:
I have read the troubleshooting and replace the module with another one without success. The problem is that it is not capturing the packages so I can not match.

sidoh, 2019-03-10:
QA on nRF24 modules sucks.  I've experienced more variance with them than any other electronics component.  I'd suggest trying another one, especially if your replacement was from the same batch.

You can verify that sends are working with a second ESP8266+nRF24 module:

https://github.com/sidoh/esp8266_milight_hub/wiki/Troubleshooting#make-sure-sends-work

^-- this is the best thing you can do to narrow down the possibilities. 

There's not a lot more I can do to help here -- the ESP8266 thinks its sending packets, which means it's communicating with the nRF24.  So you have two possibilities:

1. The nRF24 is not actually sending packets, or its signal is extremely weak (try moving the whole getup right next to the bulb).
2. The bulbs you have don't use the protocol you're trying to use them with, or are otherwise defective (do they work with an official FUT089 remote?)

A few suggestions:

* Make sure you're using a really good power supply.
* Try installing a capacitor across Gnd and Vin on the nRF24.
* If possible, try separating the ESP8266 and nRF24.  They both use 2.4 GHz, and can interfere with each other.

marcogastali2018, 2019-03-10:
Tanks.

sidoh, 2019-03-10:
Good luck!  Gonna close the issue, but happy to continue discussing over the ticket.

closed 2019-03-10 by sidoh