sidoh/esp8266_milight_hub #412: Add FUT037 Controller without remote?

coolduke666, 2019-03-10:
hi

is it possible to add a controller eg FUT037 for RGB without a remote? then you can't sniff any packages... 

because I don't need a milight controller when using MQTT :) 

how to configure the hub then? 

thx

garmck, 2019-03-10:
Have you read the wiki?
Its detailed there [https://github.com/sidoh/esp8266_milight_hub/wiki/Pairing-new-bulbs](https://github.com/sidoh/esp8266_milight_hub/wiki/Pairing-new-bulbs)

coolduke666, 2019-03-10:
sure but I dont want to pair a bulb... I want to pair a LED strip
controller.

coolduke666, 2019-03-10:
I dont want to pair a bulb... I want to pair a LED strip controller.

sidoh, 2019-03-10:
It's the same process.

sidoh, 2019-03-16:
Gonna close, @coolduke666.  Happy to answer further questions if you've got 'em.

closed 2019-04-06 by sidoh