sidoh/esp8266_milight_hub #413: Feature Request: Caching external resources

jmw6773, 2019-03-11:
I generally like to put my IOT devices on a network without Internet access, but doing so prevents the ESP Milight Hub from loading some external resources used in the webpage.

Would it be possible to have all the needed resources (JS, CSS, glyphicon fonts) residing in SPIFFS and served if the Internet version is unavailable? This could take up a bit of storage, but the files can be GZ compressed to considerably save space and decrease transfer times. (bootstrap.min.css = 107KB  --> bootstrap.min.css.gz = 19KB)

Once the required source files are in SPIFFS, you could add a check in "index.html" just after all of the scripts are loaded to check if jQuery is available. If it isn't, it can be assumed that the local cached copy of jQuery and other resources need to be used instead of the versions on the Internet.

if (typeof jQuery == 'undefined')
{
   document.write("<script src='js/jquery.min.js' type='text/javascript'</script>");
   document.write("<link rel='stylesheet' href='css/bootstrap.min.css.gz' />");
}

sidoh, 2019-03-11:
Makes sense.  This seems like a reasonable way to enable it for folks that want it without making it the default.

The polished approach would be to generate css and js bundles put them in progmem, as is currently done with `index.html`.  I've done this for other projects and it works reasonably well.  If there's enough headroom in the current image, might even make sense to make this the default.

jmw6773, 2019-03-12:
Wow, the man himself. Thank you for all of your work on this project. I built two D1 Mini controllers a few months ago in preparation for moving into my new house. I was able to put up the lights last night and play with them a little. I'm liking the newest dev builds; they seem to be much more responsive.

I just downloaded all of the jquery, bootstrap, and all of the other CSS/JS files in the index.html page. I don't know how much gulp can compress them, but they're about 600KB uncompressed. (Including a 107KB SVG file for bootloader glyphicons, which I don't think Gulp will touch.)

Even if they could be put into PROGMEM, the size of the firmware would become much larger. Do you know what the maximum size for OTA firmware updates is for the ESP8266? (I think this is why Tasmota has a minimal firmware that you have to flash before flashing an update for some firmware versions.)

It's not as polished, but another option would be to separate the firmware and the SPIFFS into separate files. On the main page, where you have the 'Update Firmware' option, you could also have an 'Upload SPIFFS" option. Then if the files are available in SPIFFS, the webpage can use them, otherwise failover to use versions from the Internet. (It might be nice to have a button to delete the files too.)

If you go this route, I would suggest having a SPIFFS_VERSION.json file in the image so the firmware can check that SPIFFS has been flashed and which version of the SPIFFS image it is using. This way you can have the firmware refuse to use the files on SPIFFS if the version is too old. (i.e. You need a newer version of jQuery than what's on SPIFFS, so fail-over to Internet version.)

I don't think SPIFFS is hardware dependent, so only one version would need to be compiled for all devices. (Just make the SPIFFS as small as needed as some devices may not have a large flash size.) 
You'd have to be careful to re-write the settings.json file again after flashing the SPIFFS though.

jmw6773, 2019-03-13:
I banged out a proof of concept yesterday and got it working. All of the files, after GZ compressing, are about 391KB. I only really need to change the index.html file and add the following to MiLightHttpServer.cpp:37
`server.serveStatic("/", SPIFFS, "/", "max-age=86400");`

When loading the page from the IP address, it took ~10-13 seconds to completely load, and when loading from DNS (milight-hub.local) it took ~45seconds. I'm not sure what's going on there, but it could be my workplace's network.

One of the issues I found is that SPIFFS has a limit of 32 characters for filenames. This is set in the SPIFFS library and would need to be changed in the firmware and in the mkspiffs utility to build the SPIFFS.bin file.

Many of the files are over this limit, including the bootstrap fonts, which means editing the bootstrap.css file to change the path to the fonts files to a shorter path. It's not too difficult, but something extra that would need to be done anytime the SPIFFS files are updated.

sidoh, 2019-03-15:
The really early versions of this project put web stuff on SPIFFS.  I don't mind making some changes that enable compatibility with this approach, but it makes setup and updates significantly more confusing if there isn't a method where everything is baked into the firmware image.  So I definitely don't want it to be the default.

> Even if they could be put into PROGMEM, the size of the firmware would become much larger. Do you know what the maximum size for OTA firmware updates is for the ESP8266? (I think this is why Tasmota has a minimal firmware that you have to flash before flashing an update for some firmware versions.)

Yeah, this would be my main concern as well.  

It depends on the firmware image.  There's some metadata that specifies the size of the SPIFFS area in flash.  I have it set to 1MB.

IIRC, for OTA to work, the new firmware image needs to be a little under half of the space reserved for the image.  So in our case, under 1.5 MB.  The latest image is 431 KB, so we've got about 1 MB to play with.  Another 500 KB would be fine; I'd be nervous about much beyond that.

> When loading the page from the IP address, it took ~10-13 seconds to completely load, and when loading from DNS (milight-hub.local) it took ~45seconds. I'm not sure what's going on there, but it could be my workplace's network.

That's pretty funky.  You've done it in what I think is the right way -- using `serveStatic`.

> One of the issues I found is that SPIFFS has a limit of 32 characters for filenames. This is set in the SPIFFS library and would need to be changed in the firmware and in the mkspiffs utility to [...]

I think the build should produce two minified and compressed artifacts: one for css, one for javascript.

Whether we put that into progmem depends on how big the resulting firmware image is.  But honestly, if it doesn't fit in progmem, it's not going to fit on SPIFFS either.  There's only 1 MB reserved, and SPIFFS is used for other stuff like config and state storage.

In all honesty, it's entirely possible that the web artifact is simply too big to fit on the ESP8266.

jmw6773, 2019-03-22:
That makes since. I think my case is just special since I don't have Internet at my house yet and am still wanting to use the Milight system. I have it coded and working on my system, but once I get Internet installed, I'll probably start using the standard build again when upgrading. 

I can put a pull request in if you want to review the changes I made, but realistically, most users won't have a need for it. 

I always thought the firmware/OTA was limited to the first 1MB of the flash and SPIFFS could use the remaining (selectable when flashing). I'm glad to hear that I'm mistaken.

I'll close this issue now unless you want me to work on it further.

closed 2019-03-22 by jmw6773

matbks, 2019-07-02:
request->send(SPIFFS, "/jquery.min.js", "text/javascript");

When this line of my code runs i have GURU MEDITATION crash, like if spiffs are not completly able to manage the jquery file.

Any help? ty