sidoh/esp8266_milight_hub #415: No connection possible

kapellmeister84, 2019-03-13:
Hey, I just tried to setup the milight hub and got stuck, after inserting my routers wifi passwort.
The status LED stoped blinking, the ESPs AP is gone and now I'm here, wondering, what to do..
I restarted the ESP several times. It just blinks once, when powered, and then nothing happens.
Maybe someone has an idea?

Thanks a lot!

kapellmeister84, 2019-03-13:
by the way - I setup the ESP with the windows flasher using firmware version 1.8.7.

JMacIV, 2019-03-13:
Try connecting back to your regular network and going to milight-hub.local to continue setup.

kapellmeister84, 2019-03-13:
that was the first thing I tried.. nothing happening

JMacIV, 2019-03-13:
Gotcha.

May want to try reflashing it. I am successfully using 1.9 dev8 build on mine. 

You can download the BIN here. https://github.com/sidoh/esp8266_milight_hub/releases <https://github.com/sidoh/esp8266_milight_hub/releases>

As a side note, I am not super savvy on these yet. If this doesn’t work, will have to wait for sidoh to chime in. 

John McMillan
(210) 863-8604

kapellmeister84, 2019-03-13:
ah, thanks man. I tried that as well, but after flashing again, not even the AP showed up.. also the status LED don't blink, as it used to do, when it waited for me to log on to its AP. Actually I had the same problem a couple days ago, with another ESP. I thought the ESP is just broken and ordered a new one (the one I am using right now). But same process - same problem..

JMacIV, 2019-03-13:
Next suggestion I think would be to download Arduino IDE or PlatformIO, then open Serial Monitor and watch what it is doing. That may help figure out where things are going wrong on the device. 

Not sure if the Windows flasher app you are using has a serial monitor or not.

kapellmeister84, 2019-03-13:
thats what serial monitor says:

pio device monitor --port /dev/cu.SLAB_USBtoUART
MBP:Milight Hub torsten$ pio device monitor --port /dev/cu.SLAB_USBtoUART
--- Miniterm on /dev/cu.SLAB_USBtoUART  9600,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

I just don't know what to do with that information

JMacIV, 2019-03-13:
It should be printing messages of what it is doing. I haven’t looked at enough of the code to see what level of detail. 

I would plug the ESP into the computer (which will restart the ESP since it didn’t have power prior or press reset button if it is already connected), IMMEDIATELY start serial monitor and watch logs. 

If it is getting weird characters showing up, the baud rate is incorrect. 

My NodeMCU chips use 115200

jmw6773, 2019-03-14:
Looking at what I have, when you first power it on, it should display that it's connecting to WiFi and what IP address it gets (or any errors). After it connects to WiFi, there aren't really any debug messages sent. 

The serial for this project uses 9600 baud rate.
This is the output I get for on my device:
```
2�ll�"$0��␇xh�␒��*WM:
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Already connected. Bailing out.
*WM: IP Address:
*WM: 192.168.56.124
Wifi connected succesfully

Setup complete (version 1.9.0-dev8-7-g34ae2c0)
```

A standard re-flash of your module with the firmware will not work as the WiFi credentials are stored in an area of memory that isn't erased by flashing. You will need to set your IDE to erase all flash if you want to re-enter your WiFi creds after a new flash.

kapellmeister84, 2019-03-14:
Thanks for your reply! So I plugged the ESP into my Mac, started the Serial Monitor and reset the ESP. This is what I got:

pio device monitor --port /dev/cu.SLAB_USBtoUART
MBP:Milight Hub torsten$ pio device monitor --port /dev/cu.SLAB_USBtoUART
--- Miniterm on /dev/cu.SLAB_USBtoUART  9600,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
␒��␀�=���

No AutoConnect, or anything..

jmw6773, 2019-03-14:
Good, that means that the firmware is running. After those messages it should give you an IP address. Try connecting from a browser on the same WiFi network to that IP.

If you still can't connect, you may need to wipe the flash to remove the WiFi credentials and try again. (I think there's a bug in WiFiManager if you have a special character in WiFi password.)

kapellmeister84, 2019-03-14:
okay, so I just erased all flash, and flashed again, with firmware version esp8266_milight_hub_nodemcuv2-1.9.0-dev8.bin
entered my wifi password (no special characters - only letters and digits) and this is what I got:

*WM: Scan done
*WM: DUP AP: Telekom_FON
*WM: DUP AP: Telekom_FON
*WM: Vodafone-BF99
*WM: -36

*WM: Vodafone Hotspot
*WM: -80
*WM: Telekom_FON
*WM: -82
*WM: Maximilian
*WM: -83
*WM: WLAN-669291
*WM: -86
*WM: abc7490
*WM: -88
*WM: WLAN-4F8QRD
*WM: -89
*WM: devolo-2e0
*WM: -92
*WM: Mr.LolliePopinson
*WM: -92
*WM: ZENTRALE
*WM: -92
*WM: Sent config page
*WM: Request redirected to captive portal
*WM: Handle root
*WM: WiFi save
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...

waited for a while.....reset the ESP:

␁��␄�␝���*WM:
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster

kapellmeister84, 2019-03-14:
just changed my wifi password to a more simple one, but got the same monitor output...

jmw6773, 2019-03-14:
I'm not sure. I don't have that much experience with the WiFiManager library.

What type of security does your WiFi network use? I use WPA2 on my 2.4GHz network and haven't had any issues.

kapellmeister84, 2019-03-14:
I use the same. I actually can see the new device in my routers setup page, but it got no IP...?
[image]

jmw6773, 2019-03-14:
Here's what I get on the first boot of my device when setting up the WiFi. (Some SSID's have been replace as they have the company name in them.)

```
--- Miniterm on COM3  9600,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
2�l��bf08��l ��*WM:
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result:
*WM: 0
*WM:
*WM: Configuring access point...
*WM: ESP3269877
*WM: milightHub
*WM: AP IP address:
*WM: 192.168.4.1
*WM: HTTP server started
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Scan done
*WM: DUP AP: ******-Public
*WM: DUP AP: ******-Public
*WM: ******-Public
*WM: -60
*WM: IT_WIFI
*WM: -73
*WM: Sent config page
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: WiFi save
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Connection result:
*WM: 3
Wifi connected successfully

Setup complete (version 1.9.0-dev8-9-g062b8ce)
```

And then after rebooting the ESP, I get the following:

```
�EO�H��|`�l :h��*WM:
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result:
*WM: 3
*WM: IP Address:
*WM: 192.168.56.124
Wifi connected successfully

Setup complete (version 1.9.0-dev8-9-g062b8ce)
```

kapellmeister84, 2019-03-14:
That looks beautiful. Can't wait to see that on my monitor 😄 

I changed my SSID, my password, tried to connect to the 5GHz Wifi (which didn't even show up in the wifi manager) - nothing worked.. it always gets stuck at "connecting as wifi client", and after reboot at "*WM: Using last saved values, should be faster"

jmw6773, 2019-03-14:
The ESP modules only support 2.4GHz, so that's why that network didn't work.

Looking though Google, another user had an issue when connecting to a Cisco AP due to the "Hw. Fragmentation Threshold" (whatever that is) being too high. His device would connect to the network, but not get an IP.

Do you have a different network (phone hotspot) that you could try testing with?

kapellmeister84, 2019-03-14:
I tried to connect it to my phone hotspot and it didn't connect. but at least the monitor told me "connection failed".  just disabled my firewall in my router - didn't change anything. still the ESP doesn't get any IP.

kapellmeister84, 2019-03-14:
tried to give the ESP a static IP but that didn't work either..

kapellmeister84, 2019-03-14:
problem solved. just connected my old wifi router and the ESP connected immediately. 
working like a charm. It seems, my new Vodafone router doesn't want the ESP in his network.

sidoh, 2019-03-14:
Thanks for helping debug, @jmw6773!  Much appreciated.

Glad you were able to resolve it, @kapellmeister84.  Gonna close.

closed 2019-03-14 by sidoh