sidoh/esp8266_milight_hub #417: Not an issue, but help needed with the Domoticz and hub.

alexsahka, 2019-03-14:
I can't figure out how is this HUB emulating a controller in the Domoticz.
Working fine with the NodeRed.

sidoh, 2019-03-15:
Hi @alexsahka, there are a couple of people who have set this up with Domoticz, but I have no experience with it myself.

cc @poudenes @joshdinsdale

If anyone is willing to contribute to the integrations documentation on the [wiki](https://github.com/sidoh/esp8266_milight_hub/wiki), it'd be very much appreciated.

poudenes, 2019-03-15:
Hi @alexsahka 

In Domoticz goto setup > Settings > Allow for 5 minutes or enable Accept new Hardware Devices
In Domoticz goto setup > Hardware
In Pull-Down at Type select: Limitless/AppLamp/Mi Light with LAN/WiFi interface
Give the remote IP address and the Port (Port is configured in the HUB > Settings > UDP)
Hit the Add button

now you have add the first DeviceID (With the 4 groups)

At the line you see SET MODE, click on it select V6 and RGBWW then Set Mode 
This setting is when you have RGBWW CCT bulbs. If you have other then Select different Type.

Now foto Setup > Devices

If everything is good you see a list with AppLamp Group1 till 4 and a All with Subtype RGBWW and also same list AppLampGroup1 till 4 and a All with Subtype RGBW.

You can only use the AppLamp Group1 till 4 and All with Subtype RGBWW.

To test the groups you can hit the bulb in front of it. Now you can see if Domoticz sends the command correct. If so then Add the Group, give it a name and DONE

Hope my English is ok to understand and how to config it.

alexsahka, 2019-03-16:
Hi poudenes,
Thank you. I got Domoticz to hub working, but not RF Remote control to Domoticz.

I can "sniff" packets in the ESP Milight HUB from remote and from Domoticz, but I can't see any responses or changes in the Domoticz Interface from the remote control. Is this normal?
This is my biggest problem, or Domoticz designed this way?

I want to control a Domoticz device with RF Remote, using ESP Milight HUB, is this possible?

Thanks.

sidoh, 2019-03-16:
It sounds like Domoticz is integrated using the UDP server, which is one-way.  The only two-way integration is MQTT.

joshdinsdale, 2019-03-16:
Hey guys

Sorry for the radio silence on this, when I get 10 mins later on I will
post my node-red flows for the remote on Domoticz. 😁

/J

sidoh, 2019-03-16:
Wonderful, thanks very much!  Would be happy to put it on the wiki. :)

joshdinsdale, 2019-03-16:
Not sure if you want my crappy code on there, it's nasty as... 😂

Will have a look when I get home in a bit. 🙂

/J

alexsahka, 2019-03-16:
This is great, thank you for sharing.

joshdinsdale, 2019-03-16:
OK dont judge me too much on this, the code is crap and functionality is basic. It just enables you to turn a channel on or off, so no dimming. That said, it all stays in state with Domoticz which is cool.

You obviously need to change it for your setup, the milight channels and the Domoticz IDX's, IP Addresses etc. 

If anyone can improve it then by all means please do! :)

`[{"id":"2f0328cf.748b18","type":"tab","label":"Milight","disabled":false,"info":""},{"id":"9529498d.3710f8","type":"mqtt in","z":"2f0328cf.748b18","name":"","topic":"milight/in/0x2621/cct/1","qos":"2","broker":"c3e2a8b7.394418","x":140,"y":80,"wires":[["48dfadf9.613774"]]},{"id":"48dfadf9.613774","type":"switch","z":"2f0328cf.748b18","name":"State","property":"payload","propertyType":"msg","rules":[{"t":"eq","v":"{\"state\":\"ON\",\"brightness\":255}","vt":"str"},{"t":"eq","v":"{\"state\":\"OFF\",\"brightness\":255}","vt":"str"}],"checkall":"true","repair":false,"outputs":2,"x":390,"y":80,"wires":[["cf511701.385e18"],["988308c3.23bb38"]]},{"id":"67cfa012.e2f8","type":"mqtt out","z":"2f0328cf.748b18","name":"","topic":"domoticz/in","qos":"","retain":"","broker":"c3e2a8b7.394418","x":776,"y":33,"wires":[]},{"id":"b77dacb3.198cf","type":"mqtt out","z":"2f0328cf.748b18","name":"","topic":"domoticz/in","qos":"","retain":"","broker":"c3e2a8b7.394418","x":775,"y":130,"wires":[]},{"id":"cf511701.385e18","type":"change","z":"2f0328cf.748b18","name":"On","rules":[{"t":"set","p":"payload","pt":"msg","to":"{\"command\": \"switchlight\", \"idx\": 489, \"switchcmd\": \"On\" }","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":544,"y":33,"wires":[["67cfa012.e2f8"]]},{"id":"988308c3.23bb38","type":"change","z":"2f0328cf.748b18","name":"Off","rules":[{"t":"set","p":"payload","pt":"msg","to":"{\"command\": \"switchlight\", \"idx\": 489, \"switchcmd\": \"Off\" }","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":544,"y":129,"wires":[["b77dacb3.198cf"]]},{"id":"99b1031a.ba63b","type":"mqtt in","z":"2f0328cf.748b18","name":"","topic":"milight/in/0x2621/cct/2","broker":"c3e2a8b7.394418","x":140,"y":240,"wires":[["63705c51.e1f594"]]},{"id":"63705c51.e1f594","type":"switch","z":"2f0328cf.748b18","name":"State","property":"payload","propertyType":"msg","rules":[{"t":"eq","v":"{\"state\":\"ON\"}","vt":"str"},{"t":"eq","v":"{\"state\":\"OFF\"}","vt":"str"}],"checkall":"true","outputs":2,"x":399,"y":252,"wires":[["1cd000a4.43b56f"],["c90ff4fa.d864f8"]]},{"id":"ef82582.37c96a8","type":"mqtt out","z":"2f0328cf.748b18","name":"","topic":"domoticz/in","qos":"","retain":"","broker":"c3e2a8b7.394418","x":773,"y":203,"wires":[]},{"id":"dab187a7.881f88","type":"mqtt out","z":"2f0328cf.748b18","name":"","topic":"domoticz/in","qos":"","retain":"","broker":"c3e2a8b7.394418","x":772,"y":300,"wires":[]},{"id":"1cd000a4.43b56f","type":"change","z":"2f0328cf.748b18","name":"On","rules":[{"t":"set","p":"payload","pt":"msg","to":"{\"command\": \"switchlight\", \"idx\": 490, \"switchcmd\": \"On\" }","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":541,"y":203,"wires":[["ef82582.37c96a8"]]},{"id":"c90ff4fa.d864f8","type":"change","z":"2f0328cf.748b18","name":"Off","rules":[{"t":"set","p":"payload","pt":"msg","to":"{\"command\": \"switchlight\", \"idx\": 490, \"switchcmd\": \"Off\" }","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":541,"y":299,"wires":[["dab187a7.881f88"]]},{"id":"545c5cd4.4049f4","type":"mqtt in","z":"2f0328cf.748b18","name":"","topic":"milight/in/0x2621/cct/0","qos":"2","broker":"c3e2a8b7.394418","x":140,"y":420,"wires":[["3d96d58d.29e8fa"]]},{"id":"3d96d58d.29e8fa","type":"switch","z":"2f0328cf.748b18","name":"State","property":"payload","propertyType":"msg","rules":[{"t":"eq","v":"{\"state\":\"ON\"}","vt":"str"},{"t":"eq","v":"{\"state\":\"OFF\"}","vt":"str"}],"checkall":"true","outputs":2,"x":399,"y":432,"wires":[["3833f17c.e7609e"],["d914e20b.0cb68"]]},{"id":"6ebf42fb.5d38dc","type":"mqtt out","z":"2f0328cf.748b18","name":"","topic":"domoticz/in","qos":"","retain":"","broker":"c3e2a8b7.394418","x":773,"y":383,"wires":[]},{"id":"6884f553.88296c","type":"mqtt out","z":"2f0328cf.748b18","name":"","topic":"domoticz/in","qos":"","retain":"","broker":"c3e2a8b7.394418","x":772,"y":480,"wires":[]},{"id":"3833f17c.e7609e","type":"change","z":"2f0328cf.748b18","name":"On","rules":[{"t":"set","p":"payload","pt":"msg","to":"{\"command\": \"switchlight\", \"idx\": 3, \"switchcmd\": \"On\" }","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":541,"y":383,"wires":[["6ebf42fb.5d38dc"]]},{"id":"d914e20b.0cb68","type":"change","z":"2f0328cf.748b18","name":"Off","rules":[{"t":"set","p":"payload","pt":"msg","to":"{\"command\": \"switchlight\", \"idx\": 3, \"switchcmd\": \"Off\" }","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":541,"y":479,"wires":[["6884f553.88296c"]]},{"id":"c3e2a8b7.394418","type":"mqtt-broker","z":"2f0328cf.748b18","name":"","broker":"127.0.0.1","port":"1883","clientid":"","usetls":false,"verifyservercert":true,"compatmode":true,"keepalive":"60","cleansession":true,"birthTopic":"","birthQos":"0","birthRetain":null,"birthPayload":"","closeTopic":"","closePayload":"","willTopic":"","willQos":"0","willRetain":null,"willPayload":""}]`

alexsahka, 2019-03-16:
This flow doesn't work correctly, I have to make few adjustments (not MQTT user/pass or other typical stuff) but when it starts working it creates a loop MQTT IN to OUT if Domoticz out connected over UDP to ESP Milight HUB  (obvious).

So,
Currently, it is working only one way:
1. Domoticz out to ESP Milight HUB working OK.
2. Remote control -(RF)- to ESP Milight HUB -(MQTT)- NodeRed -(MQTT)- Domoticz only ON/OFF. 
3. I can add some code to Node Red flow and brightness will work. Not a big deal.

I was able to reproduce this even before my original question.

I think the biggest problem is that ESP Milight HUB (out command) cannot communicate directly with Domoticz over UDP for some reasons.

 In my opinion, it should be:
 Remote control -(RF)- to ESP Milight HUB -(UDP)- Domoticz.

sidoh, 2019-03-16:
UDP is and always will be one way, in the direction of Domoticz -> espMH.

The protocol is reverse engineered from Milight's official wifi gateways, and was not designed to be two-way.

joshdinsdale, 2019-03-16:
Yes I use the UDP gateway with it as well to achieve a reasonable resuly.
But as I said, although it's messy it's been very reliable.

alexsahka, 2019-03-16:
OK I see, for some reasons, I thought that UDP works two way from ESP Milight HUB.
Spent countless hours to "sniff" UDP traffic from ESP Milight HUB, even install Wireshark for this purpose.

sidoh, 2019-03-16:
The UDPv6 protocol sends data back as acknowledgements, but there's no way for espMH to communicate back to the other end "hey, i saw this packet."  This makes sense if you think about the purpose the official Milight wifi gateways serve.

MQTT _is_ two way.  You can subscribe to a state topic which is updated when packets are intercepted.

sidoh, 2019-04-06:
@alexsahka, anything else here?  Will close the issue if not.

@poudenes, thank you very much for the instructions.  I'd like to include some documentation in the wiki.  Let me know if you want to make any edits to what you have here.

alexsahka, 2019-04-08:
The question is still existed, how to fully emulating a two-way controller in the Domoticz (without NodeRed).

sidoh, 2019-04-08:
Got it.

The MQTT integration does what you want.  I don't know enough about Domoticz to help beyond that.

joshdinsdale, 2019-04-08:
A plugin would be ideal, in the same way they have made one for
zigbee2mqtt, however it's not really Chris's responsibility and I wouldn't
know where to start!

/J

poudenes, 2019-04-08:
> The question is still exist, how to fully emulating a two-way controller in the Domoticz (without NodeRed).

Domoticz itself cant send MQTT from a device to something. I create some workarounds that works. 
But what do you want?

1) turn on a Bulb direct from the Hub and send a signal to Domoticz to turn bulb on as well without action.
2) turn on a bulb from Domoticz and send a signal to the Hub without any other action.

Tell me :)

alexsahka, 2019-04-12:
Tell me :)
 A simple working solution for Domoticz, similar to this:
https://github.com/sidoh/esp8266_milight_hub/wiki/Using-Milight-Remote-with-HomeAssistant

Thanks, this is my original question.
Turning Lights On/Off this is not a big deal, need full functional control of color, tone e.t.c.
..... If it's technically possible, if not, then I'll switch to HA instead of Domoticz.

galinette2000, 2019-05-02:
Dear all,
I have implemented a Domoticz plugin for this hub, which uses MQTT and is fully bidirectional. Controls color, tone, on/off, disco mode, etc...
Please feel free to test it! I open a dedicated issue about this

poudenes, 2019-05-02:
Where can we find it :D?

sidoh, 2019-05-02:
https://github.com/sidoh/esp8266_milight_hub/issues/452

sidoh, 2019-05-19:
Instructions added to the wiki [here](https://github.com/sidoh/esp8266_milight_hub/wiki/Domoticz).

Thanks @galinette2000 for the plugin!
Thanks @poudnes for the documentation!

closed 2019-05-19 by sidoh