sidoh/esp8266_milight_hub #418: MQTT Connection status

starwolf73, 2019-03-15:
Hi,
Using Home Assistant HASSIO and MQTT pluging, I'm trying to find a way to trigger when Milight_hub connect/reconnect to the MQTT server to assigned an automation task.  

Is it possible for the device to announce himself in a specific topic? Unless there is already a feature in MQTT Mosquitto I didn't saw.

Thanks
Eric

sidoh, 2019-03-16:
Hi Eric,

The 1.9.0 dev branches support MQTT LWT, which sends a message to a specified topic when the device disconnects.

There is not currently a way to send a message when first connecting, but that wouldn't be terribly hard to add.

starwolf73, 2019-03-21:
Thanks @sidoh. I believe the feature is call MQTT Birth. [https://www.home-assistant.io/docs/mqtt/birth_will/](url). I'll rename the title for sake of clarity.

sidoh, 2019-03-24:
This is released in [1.9.0-dev10](https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.9.0-dev10).

starwolf73, 2019-03-24:
Thanks, I'll give a try this week and provide feedback.

rejoe2, 2019-04-12:
E.g. tasmota by default uses the LWT to inform about the online and offline status. I personally found this helpfull, but afai can see, there's no option to configure a "opposit to LWT" message. 

So is there an option to add also a online LWT message when the hub is connecting to the broker to revoke e.g. "offline"?

sidoh, 2019-04-12:
Yes, it's in the 1.9.0 branch, as mentioned above.  Try the RC release:

https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.9.0-rc2

rejoe2, 2019-04-13:
My remark already refferenced to version 1.9.0-rc2. Both message types (LWT and birth messages) are great! But one has to "clean up" the LWT message "manually" on the HA controller side (may be different in HASS), as there is only one message to the LWT topic, but not a second, indicating the hub is online again. This one (birth) is to a different topic and in a different structure - single word payload (or whatever is configured) vs. JSON. 
So my suggestion would be to also send a second "birth" message info to LWT also (same way as eg. tasmota devices do).

sidoh, 2019-04-13:
Okay, I understand the suggestion now.  But not sure I see the value.  Why should it be the same topic?  Why should it have the same structure?  Is this some standard that I'm not aware of?  If it's a broad thing that many MQTT integrations do, happy to consider it.  If it's just a particular arbitrary way of doing things, I think I'll probably pass.

rejoe2, 2019-04-13:
Same topic, same structure: My HA software expects the same type (JSON/non-JSON), otherwise I have to add a more complex message treatment (which is possible).
Looking at the IBM official guide at https://www.ibm.com/support/knowledgecenter/en/SSFKSJ_7.1.0/com.ibm.mq.doc/tt60360_.htm there's at the end of the text:
> To monitor connections, complement the "last will and testament" publication with other publications to record connections and programmed disconnections. 
So imo the tasmota implementation - just sending "Online" and "Offline" seems to be very close to the inventor's description of the intended behaviour. On the other hand, at least imo, there's no need to make the LWT message configurably at all, if, an option to manipulate the time before LWT (when disconneted) is sent used by the broker.

sidoh, 2019-04-14:
Ah ok, I see the rationale.  It's a single topic that indicates device status.  That is nicer.

What if I change this to have a single topic used for LWT, birth, disconnects, etc.?  It could have all of the same information that the birth message does (with IP address, firmware version, etc.), and also include a "status" field which could be one of `connected`, `disconnected_gracefully`, or `disconnected_ungracefully`. 

How does this sound?

rejoe2, 2019-04-15:
No need for end-users to configure things twice, plus additional info privided. 
Imo that sound great!

sidoh, 2019-04-15:
Live in 1.9.0-rc3:

https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.9.0-rc3

sidoh, 2019-04-15:
Thank you for the suggestion!

rejoe2, 2019-04-15:
Thanks a lot for implemeting that fast! 
V. 1.9.0-rc3 is up an running *grin*

lightning1, 2019-05-19:
This looks quiet nice. But sadly, Homeassistant does not support extracting values from json when it comes to availability. To use this with HA flawlessly, a predictable payload to a fixed topic is required (e.g. "ON" or "OFF").
See [Homeassistant MQTT Light documentation](https://www.home-assistant.io/components/light.mqtt/) for details. Lookout for the settings:
- availability_topic
- payload_available
- payload_not_available

sidoh, 2019-05-19:
Yeah, I noticed this as well.  It probably should support templates for this, but I don't have the bandwidth to open a PR for it right now.

I'll add a setting to support simple client status messages (will just hardcode "connected" and "disconnected").

lightning1, 2019-05-21:
Great, this work out quiet well.
Thank you, keep up the good work!

closed 2019-05-28 by sidoh