sidoh/esp8266_milight_hub #419: Group 0 updates missing bulb_mode changes

zyrorl, 2019-03-15:
It appears that when you use a remote to change group 0 to trigger a change from white mode to colour mode and vice versa that while hue and saturation values are sent, the bulb_mode is not, which means that you don't know what mode the bulbs have all been set to.

sidoh, 2019-03-16:
Yes, that's expected.

Think of the updates topic as a pretty primitive translation of the raw packets being sent to the bulbs: "these bytes mean the command X was sent to the bulb Y."

`bulb_mode` is a derived state field, and is not a command of its own.  You can infer `bulb_mode` from other commands.  For example, if you see a `hue` command, you know the state has changed to `color`.

sidoh, 2019-03-16:
group 0 does not usually behave like you want it to w.r.t. to state.  have an idea for how to improve this, but still messing around with it.

Gonna close since this is expected behavior. lmk if i've misunderstood.

closed 2019-03-16 by sidoh