sidoh/esp8266_milight_hub #423: FUT098 FUT025 BRIGHTNESS

flokycool, 2019-03-24:
Hi,

I have a FUT025 (with a FUT098 remote). Beside the Issue that the Remote messages are not seen by the HUB (Packets from some remotes (e.g., RGB) are never sniffed) #266  
i have a issue with the **Brightness**. No matter when I go Lower or Higher it is only sendig High. I tried it over the Web Interface, MQTT and the UDP Protocol. The Issue is the same.

I did a Sniff on the Hub and could see that it is sending always 0x03 no matter if I go up or down on the Ruler.

sidoh, 2019-03-24:
Almost positive this is expected behavior.

FUT098 only sends up/down commands, which are relative.  There's no way to know what the brightness started at.  If you send enough ups/downs, it'll know it's at min/max brightness and an absolute reference point will be stored, and you'll see brightness updates.

sidoh, 2019-04-06:
@flokycool, please let me know if I've misunderstood, and we can re-open the issue.

closed 2019-04-06 by sidoh