sidoh/esp8266_milight_hub #427: Need some assistance with espMH receiving commands on MQTT

SyntaxArch, 2019-04-06:
Using the latest pre-release 1.9.0-dev12 release. Had the same problem with 1.9.0-dev10

I'm trying to get the hub to receive commands via mqtt and execute the corresponding command on the specified DeviceID/BulbType/Group. 

Everything else works fine. Using the WebUI to send commands, etc.

I am unable to get the hub to receive/respond/acknowledge the mqtt command sent to 'mqtt_topic_pattern' which is 'cmnd/milighthub/CommonMiLightHub/:hex_device_id/:device_type/:group_id'

I'm using an inject node on mqtt with the command :
`{
    "status": true,
    "color": {
        "r": 255,
        "g": 200,
        "b": 255
    },
    "level": 100
}`

sent to the topic: milighthub/CommonMiLightHub/0x1/fut089/0

Below is my settings.json for the hub:

`{"admin_username":"","admin_password":"","ce_pin":4,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":20,"http_repeat_factor":2,"auto_restart_period":0,"mqtt_server":"192.168.1.10","mqtt_username":"CommonMiLightHub","mqtt_password":"CommonMiLightHubPassword","mqtt_topic_pattern":"cmnd/milighthub/CommonMiLightHub/:hex_device_id/:device_type/:group_id","mqtt_update_topic_pattern":"updates/milighthub/CommonMiLightHub/:hex_device_id/:device_type/:group_id","mqtt_state_topic_pattern":"status/milighthub/CommonMiLightHub/:hex_device_id/:device_type/:group_id","mqtt_lwt_topic":"","mqtt_lwt_message":"","mqtt_birth_topic":"","discovery_port":48899,"listen_repeats":2,"state_flush_interval":200,"mqtt_state_rate_limit":100,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":20,"packet_repeat_minimum":4,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Off","led_mode_packet":"Flicker","led_mode_packet_count":3,"hostname":"CommonMiLightHub","rf24_power_level":"MAX","rf24_listen_channel":"LOW","wifi_static_ip":"","wifi_static_ip_gateway":"","wifi_static_ip_netmask":"","rf24_channels":["LOW","MID","HIGH"],"device_ids":[1,2,16],"group_state_fields":["state","status","brightness","level","hue","saturation","color","mode","kelvin","color_temp","bulb_mode","computed_color","effect","device_id","group_id","device_type","oh_color"]}`

Are there any logs which I can see? Because there is noting being published on the status topic for the above case. Would highly appreciate any assistance. I'm not sure what/where I went wrong. It could be a pretty simple/silly mistake on my part. I haven't been able to figure out what it is though.

Please let me know if any more information about my setup is required to identify the issue.

sidoh, 2019-04-06:
Settings look fine.  

What topic are you listening on?  Should be `status/milighthub/CommonMiLightHub/+/+/+`

Does the bridge respond in other ways to commands being sent to it (e.g., does it control bulbs)?

sidoh, 2019-04-06:
You can enable MQTT Serial logging if you recompile with the `MQTT_DEBUG` flag:

https://github.com/sidoh/esp8266_milight_hub/blob/master/platformio.ini#L29

chietanjain, 2019-04-07:
Thanks for the response @sidoh . I am working with @SyntaxArch  on this project.

Yes, all other functions work fine.

Status Topic and update topic get messages too when receiving packets from the MiLight Bridge 3.0, but not when a status/command is sent to cmnd/milighthub/CommonMiLightHub/0x1/fut089/0 (or for any other device id, type and group).

We will enable serial logging and revert.

JMacIV, 2019-04-07:
@sidoh - Is the hex device ID too short?

@chietanjain - I haven’t used Chris’s project with the fut08 bulbs before, but with the RGBCCT bulbs, the hex device id in my setup are “0x1B0B” and “0x1B0C”. 

My B if this is not necessary. Just thought I’d chime in since I noticed that.

chietanjain, 2019-04-07:
Enabled `MQTT_DEBUG`:

No serial logs when publishing to: 
```
cmnd/milighthub/CommonMiLightHub/0x1/fut089/0
```
But when using the web interface it logs as expected:
```
MqttClient - publishing update to updates/milighthub/CommonMiLightHub/0x1/fut089/0
MqttClient - publishing update to status/milighthub/CommonMiLightHub/0x1/fut089/0
```
I have checked and rechecked for typos, hoping I find one, but there are none.

The MQTT Client on the bridge doesn't seem to be listening on the **MQTT topic pattern** which is set to:
`cmnd/milighthub/CommonMiLightHub/:hex_device_id/:device_type/:group_id`

What am i doing wrong?!

chietanjain, 2019-04-07:
@JMacIV , Thanks. I tried with longer device IDs. Still the same !

JMacIV, 2019-04-07:
> @JMacIV , Thanks. I tried with longer device IDs. Still the same !

Wish that would have helped! I'm not a dev of any sort, but try to make suggestions if I have one since Chris already does a ton on this project!

I did just notice this from your original post (emphasis added), could just be a typo, but figured I would mention. See below.

> I am unable to get the hub to receive/respond/acknowledge the mqtt command sent to 'mqtt_topic_pattern' **which is 'cmnd/milighthub/CommonMiLightHub/:hex_device_id/:device_type/:group_id'**
>
> I'm [...]

Your mention of "sent to the topic: milighthub/CommonMiLightHub/0x1/fut089/0" does not match your mqtt_topic_pattern string above. It is missing the leading 'cmd/'

chietanjain, 2019-04-07:
@JMacIV, thanks very much for pointing that out.

Unfortunately this is a typo in the post and not in the actual inject node on nodered.

chietanjain, 2019-04-07:
update: I tried changing the payload from `JSON String` to `String `on NodeRed, still the same (i.e.) no status update from espMH and injected commands are not being being sent to the connected lights.

Here is the debug output from NodeRed when sending the payload as a `string`. Sharing it in case I am missing something:
```

object
topic: "cmnd/milighthub/CommonMiLightHub/0x1/fut089/0"
payload: "{"status":"ON","color":{"r":255,"g":200,"b":255},"level":10}"
qos: 0
retain: false
_topic: "cmnd/milighthub/CommonMiLightHub/0x1/fut089/0"
_msgid: "4040d2bf.c1bf4c"

object
topic: "cmnd/milighthub/CommonMiLightHub/0x1/fut089/0"
payload: "{"status":"OFF"}"
qos: 0
retain: false
_topic: "cmnd/milighthub/CommonMiLightHub/0x1/fut089/0"
_msgid: "79f99ac8.4ab004"
```

chietanjain, 2019-04-07:
The issue was that I was using `:hex_device_id` instead of `:device_id`. I should have followed the tool tip help instead of assuming that `:hex_device_id` will also work.

Enabling MQTT Debug serial log helped.

sidoh, 2019-04-07:
Ah, I should've caught that.  `:device_id` should support both hex and dec (it just looks for the `0x` prefix for hex).

It's confusing that updates/state support those two tokens, but commands doesn't.  I'll look into fixing.

sidoh, 2019-04-07:
Thanks very much @JMacIV for helping with the troubleshooting process. :)

sidoh, 2019-04-07:
Going to close this issue since it seems like you've got it figured out, and the source of confusion will be fixed in 1.9.0 (released in 1.9.0-rc2).

closed 2019-04-07 by sidoh