sidoh/esp8266_milight_hub #429: Rest-API broken in v1.9.x?

bombcheck, 2019-04-07:
Since v1.9.x I am unable to switch my RGBCCT-Bulbs via the REST-API. v6-UDP-Servers still work fine...

sidoh, 2019-04-07:
Seems to be working for me on 1.9.0-rc1.  I've also tested the REST API for each dev release.

Example command for me --

```
$ curl -X PUT -d '{"status":"on"}' -H 'content-type: application/json' milight-hub/gateways/0x118D/rgb_cct/1
{"status":"OFF","kelvin":100,"bulb_mode":"white","effect":"white_mode"}%
```

Can you provide a command that reproduces the issue?  What version are you upgrading from?

bombcheck, 2019-04-08:
I did a clean flash to wipe out any issues with older versions... Will check my hardware this evening. Maybe a faulty transceiver... But with 1.8.x it works fine. Strange...

In your example above you set "status:on" but it returns "status:off"?

My system for example sends these command when I turn a bulb on:
{"status":"on"}
{"status":"on","command":"set_white"}
{"level":'.50',"temperature":'50'}

Nothing has been changed on my software. Above commands work on v1.8.x, but fail on v1.9.x

sidoh, 2019-04-08:
Ah, that was just a lazy copy/paste error.  It seems to work fine for me:

```
$ curl -X PUT -H 'Content-Type: application/json' -d '{"status":"off"}' 10.133.8.158/gateways/0x2200/rgb_cct/1
{"state":"OFF","brightness":26,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}%

$ curl -X PUT -H 'Content-Type: application/json' -d '{"status":"on"}' 10.133.8.158/gateways/0x2200/rgb_cct/1
{"state":"ON","brightness":26,"color_temp":370,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}%

$ curl -X PUT -H 'Content-Type: application/json' -d '{"level":50,"temperature":50}' 10.133.8.158/gateways/0x2200/rgb_cct/1
{"state":"ON","brightness":128,"color_temp":262,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}%
```

bombcheck, 2019-04-08:
Maybe because I run the curl request without any delay or something?

sidoh, 2019-04-08:
Does reverting to 1.8.x fix things?  If so, does an older version of 1.9.0 work?  Maybe 1.9.0-dev11?

bombcheck, 2019-04-08:
Yes 1.8.x work fine. Habe not tried other 1.9

sidoh, 2019-04-08:
Don't think it's a timing thing.  If I nuke state and spam commands one after the other, still looks fine:

```
$ curl -X DELETE -H 'Content-Type: application/json' 10.133.8.158/gateways/0x2200/rgb_cct/1
true%

$ curl -X PUT -H 'Content-Type: application/json' -d '{"status":"off"}' 10.133.8.158/gateways/0x2200/rgb_cct/1 && echo && curl -X PUT -H 'Content-Type: application/json' -d '{"status":"on"}' 10.133.8.158/gateways/0x2200/rgb_cct/1 && echo && curl -X PUT -H 'Content-Type: application/json' -d '{"command":"set_white"}' 10.133.8.158/gateways/0x2200/rgb_cct/1 && echo && curl -X PUT -H 'Content-Type: application/json' -d '{"level":50,"temperature":50}' 10.133.8.158/gateways/0x2200/rgb_cct/1
{"state":"OFF","color":{"r":255,"g":255,"b":255}}
{"state":"ON","color":{"r":255,"g":255,"b":255}}
{"state":"ON","color_temp":153,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}
{"state":"ON","brightness":128,"color_temp":262,"bulb_mode":"white","color":{"r":255,"g":255,"b":255},"effect":"white_mode"}%
```

bombcheck, 2019-04-08:
Really strange... Changes power converter for my tranceiver and also changed CE pin. Will test again.

bombcheck, 2019-04-09:
Maybe because I compile myself? Always used my mac for that, but this time tried under windows...
Mac gives me strange errors when trying to compile...

sidoh, 2019-04-09:
> Really strange... Changes power converter for my tranceiver and also changed CE pin. Will test again.

If reverting to 1.8 fixes the problem, it's obviously a software thing.  Not sure if that's what you meant.

> Maybe because I compile myself? Always used my mac for that, but this time tried under windows...

Ah yeah, possibly.  I've seen Windows do funky things (see this issue comment https://github.com/sidoh/esp8266_milight_hub/issues/109#issuecomment-335196173).  I reverted this change recently actually because I couldn't remember why I was running a fork of nRF24, and I was tired of maintaining forks of libraries.  Now I remember why 😅 

The change to try would be to change this line:

https://github.com/sidoh/esp8266_milight_hub/blob/1.9.0/platformio.ini#L19

to

```
  sidoh/RF24
```

> Mac gives me strange errors when trying to compile...

I remember that you had opened another issue for this.  I always use a Mac, and have never seen those errors.

There are several issues on the PlatformIO project with the sort of errors you're seeing.  Maybe check those out.  Don't think it's an issue with this project.

example:

https://github.com/platformio/platformio-atom-ide/issues/1448

bombcheck, 2019-04-09:
Strange... Now it does compile on my mac without issues. And without changing anything...
But: Issue on 1.9.x persists: My RGBCCT-bulbs do respond, but very few times. On 1.8.x it works way more reliable.

sidoh, 2019-04-09:
Definitely a PlatformIO/system Python thing.  I can't help solve that mystery. :)

Please try the above suggestions.  Also please provide the output of `/settings` and Serial logs.

bombcheck, 2019-04-10:
Ok, will try to compile with sidoh/RF24 on my mac later...

sidoh, 2019-04-11:
awesome, thanks.  If it works at all (but only intermittently), it's probably not the RF24 thing, actually.  The issue there caused a crash.

Serial logs (and output of `/settings`) are probably what we need.

bombcheck, 2019-04-13:
OK. Settings:

> {"admin_username":"admin","admin_password":"nb230901","ce_pin":4,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":25,"http_repeat_factor":3,"auto_restart_period":0,"mqtt_server":"","mqtt_username":"","mqtt_password":"","mqtt_topic_pattern":"","mqtt_update_topic_pattern":"","mqtt_state_topic_pattern":"","mqtt_lwt_topic":"","mqtt_lwt_message":"","mqtt_birth_topic":"","discovery_port":48899,"listen_repeats":0,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":150,"packet_repeat_minimum":3,"enable_automatic_mode_switching":true,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Off","led_mode_packet":"Flicker","led_mode_packet_count":3,"hostname":"milight-hub","rf24_power_level":"MAX","rf24_listen_channel":"LOW","wifi_static_ip":"","wifi_static_ip_gateway":"","wifi_static_ip_netmask":"","rf24_channels":["LOW","MID","HIGH"],"device_ids":[6660,1827,37677,48972,61770],"gateway_configs":[[48972,8901,5],[37677,8902,5],[61770,8903,5],[1827,8904,5]],"group_state_fields":["state","brightness","color","mode","color_temp","bulb_mode"]}

Will post serial logs in a few minutes...

bombcheck, 2019-04-13:
Nothing except the boot log on serial...

sidoh, 2019-04-14:
Got it, interesting.  And you said UDP worked without issue?

Can you copy/paste some terminal output similar to what I've got in this comment that reproduces the issue?

https://github.com/sidoh/esp8266_milight_hub/issues/429#issuecomment-480929667

sidoh, 2019-04-19:
Any update here, @bombcheck?  Don't want to release 1.9.0 if there's some sort of bug it introduced.  But having a hard time imagining any of the features introducing this behavior.

bombcheck, 2019-04-20:
It is really strange. Something must have changed... Maybe I should walk through the commits to check...
But currently I am on vacation with my family.

sidoh, 2019-04-20:
No rush.  Let me know when you have a chance to check.  Have a nice vacation!

sidoh, 2019-05-27:
Hey @bombcheck, just checking in here.  Let me know if you're still having this issue.  Would like to close it if not.

sidoh, 2019-06-07:
Gonna close this since I haven't heard back, but happy to re-open upon request.

closed 2019-06-07 by sidoh

nanokatz, 2019-11-16:
I am also having this issue on a nodemcuv2 install from the released bin file. Previous version worked fine.
Calling http://milight-hub/gateways/0xF746/rgb_cct/1
gets me just 

  | {"success":true}
-- | --

Boot log shows nothing.

sidoh, 2019-11-16:
This does not sound like the same issue.

Append `?blockOnQueue=true` and it'll work.  The GET request _should_ be returning state without that parameter, but it doesn't right now.